skiffos/SkiffOS #15: When compiled as a .iso image and started as a VM start script fails

ohadln, 2017-05-08:
Start script cannot find scratchbuild.bash, thus the core docker container cannot be loaded and initialization fails.
Compiled as intel x64 with the following file(configs/intel/x64/buildroot/iso):

BR2_TARGET_ROOTFS_ISO9660=y
BR2_TARGET_GRUB=y

paralin, 2017-05-08:
Scrach build is provided here: https://github.com/paralin/scratchbuild/blob/master/scratchbuild.bash

The script is not actually useful on x86 since most Docker images on the Docker Hub are built against x86. The script probably should be modified to do nothing against x86.

I'm planning to replicate your setup tonight or tomorrow. In the meantime, you can bypass this entire problem by running the following command on your running system:

```
printf "#!/bin/bash\ndocker pull $2" > /opt/skiff/scripts/scratchbuild.bash
```

Please let me know if that fixes your issue for now.

ohadln, 2017-05-11:
Hi, the workaround is not possible, as the VM does not reach a prompt at all, see attached screenshot.

[image]

paralin, 2017-05-11:
Are you certain? Try SSHing to root. At that stage it should have already completed boot. You're just watching the system logs.

paralin, 2017-05-15:
You are setting BR2_TARGET_GRUB but BR2_TARGET_GRUB2 is already enabled - do you need both? Why?

ohadln, 2017-05-18:
Hi Christian, I don't need both - just GRUB. I'm not sure though how to disable GRUB2.

paralin, 2017-05-18:
`BR2_PACKAGE_GRUB2=n`

paralin, 2017-05-28:
Superseded: #26

closed 2017-05-28 by paralin