skiffos/SkiffOS #25: How to change the console options

ohadln, 2017-05-21:
Hi, working with SkiffOS in a VM(VMWare / Virtualbox) I want to redirect the console to a file on the host. How do I change the console options?

Thanks in advance,
Ohad

paralin, 2017-05-21:
Hi @ohadln

The system uses systemd-journald at the moment. You can customise this using the standard systemd-journald config files in a rpot

paralin, 2017-05-23:
If you actually want to redirect the entire console, you need to change the kernel arguments and add a `console=/dev/myTTY0,115200`, if you're using Grub this will be in the grub config.

ohadln, 2017-05-26:
Hi Christian, 

Thanks Christian, I know about this option, however, not sure where to put it. Is there an option in the kernel for that or do I add it with root_overlay or post_build script(replacing /boot/grub/grub.cfg)?

Thanks,
Ohad

paralin, 2017-05-26:
How are you making the grub cfg and installing grub right now? I'll bundle all of this into an example.

paralin, 2017-05-29:
Am happy to help here with this more, I'm working on a vagrant example right now. Feel free to comment if you want me to re-open.

ohadln, 2017-06-01:
Added the following inside configs/intel/x64/kernel/grub:

GRUB_CMDLINE_LINUX_DEFAULT="quiet" 
GRUB_CMDLINE_LINUX="console=tty0 console=ttyS0,115200"
GRUB_TERMINAL=serial console
GRUB_SERIAL_COMMAND="serial --unit=0 --speed=115200 --stop=1"

However, doesn't see to output the console logs to the serial console(serial console configured on VirtualBox and redirected to a file).

paralin, 2017-06-02:
I'll try to reproduce this when working on the vagrant example. I've been busy re-writing skiff-core in the last week or so. By the way - if you're using it, let me know what you think of the new version, it's a complete rewrite.

paralin, 2017-06-05:
@ohadln configs/intel/x64/kernel/grub - the "kernel" directory inside a config is intended for kernel config fragments, that is for the kernel. Afaik specifying GRUB configurations here won't do anything. You need to override wherever you generate the grub.cfg.

closed 2017-08-13 by paralin