skiffos/SkiffOS #27: Create Image File on Disk

Compy, 2017-05-28:
Is there any way to create an image file locally on disk that can be flashed on another machine? I tend to do these builds in a VM and then flash them onto a separate device as an img file.

I am targeting the odroid XU4 if that matters any.

Thank you!

paralin, 2017-05-28:
Yes, this is possible with genimg, I'll make a command for the odroids to build an image.

paralin, 2017-05-28:
Should be done within the next ~5 hours or so.

paralin, 2017-05-29:
Done, please try this out and let me know if it works.

 1. Pull the latest master version of Skiff.
 2. Rebuild Skiff (with `make clean` and then re-build).
 3. Build the OS image with `make cmd/odroid/common/buildimage`
 4. Flash the OS image to your storage media with either http://etcher.io or `dd if=sdcard.img of=/dev/sdX bs=1M`

The OS may not boot, in which case please pull up `gparted` or similar and check to make sure the partitions and filesystems look OK, as well as the files on `Boot` (should be .dtb, .cpio.uboot, boot.scr, zImage). If it doesn't boot with all of this in place, it's probably because U-Boot is not flashing right. I'll look into this if so.

paralin, 2017-06-15:
Closing, please open an issue if the script doesn't work correctly.

closed 2017-06-15 by paralin