skiffos/SkiffOS #34: Docker port forward on SkiffOS

ohadln, 2017-06-07:
Tried to run a container with -p options, then get the following error:

# docker run -d -p 6001:6001 alpine
c2c8679cee4a95551610162ad4a48e6cf8447adff1d5ab089c215aa927bf5d0b
docker: Error response from daemon: driver failed programming external connectivity on endpoint wizardly_golick (4b318ebe3490ff09c5256a8ee88a23bde0ba1f226edc136226f8a62099dec727): exec: "docker-proxy": executable file not found in $PATH.

It seems that docker-proxy is missing on the machine. How do I add docker-proxy?

paralin, 2017-06-07:
Good call, this is actually a bug in Buildroot... Investigating.

paralin, 2017-06-07:
Fixed, and relevant patches submitted to upstream:

 - https://patchwork.ozlabs.org/patch/772729/
 - https://patchwork.ozlabs.org/patch/772728/

Let me know if you still experience this issue after:

```
git pull
git submodule update --init --recursive
make compile
```

Buildroot now includes `docker-proxy` binary in the target.

commit 5a7aaff3 (2017-06-07): fix(buildroot): add docker-proxy to buildroot

Fixes #34
closed 2017-06-07 by paralin