skiffos/SkiffOS #45: build for pi3 fails on skiff-core

ohadln, 2017-08-16:
Trying to build SkiffOS for pi3. The build fails with the following error messages:
$ SKIFF_CONFIG=pi/3,skiff/core SKIFF_WORKSPACE=pi3 make configure
$ SKIFF_WORKSPACE=pi3 make compile
*
*
*
package/pkg-generic.mk:216: recipe for target '/home/tadmin/work/SkiffOS/workspaces/pi3/build/skiff-core-v0.3.1/.stamp_built' failed
make[3]: *** [/home/tadmin/work/SkiffOS/workspaces/pi3/build/skiff-core-v0.3.1/.stamp_built] Error 2
Makefile:16: recipe for target '_all' failed
make[2]: *** [_all] Error 2
make[2]: Leaving directory '/home/tadmin/work/SkiffOS/workspaces/pi3'
Makefile:52: recipe for target 'compile' failed
make[1]: *** [compile] Error 2
make[1]: Leaving directory '/home/tadmin/work/SkiffOS/build'
Makefile:2: recipe for target 'compile' failed
make: *** [compile] Error 2

paralin, 2017-08-16:
Can you include the logs earlier up in the build? The part you pasted doesn't have the actual build error. Thanks!

paralin, 2017-09-06:
Closing due to lack of response. Builds fine for me - if it's still broken, try cleaning and rebuilding

justin0a0, 2017-10-01:
I have been trying to build for an Odroid-XU4 with the same issue.
[Log](https://gist.github.com/justin0a0/7e0e9a97c6e3b0c97eb1131a2c1af613)

paralin, 2017-10-01:
I'll fix that in the next 20min or so.

paralin, 2017-10-03:
@justin0a0 fixed.

closed 2017-10-03 by paralin

justin0a0, 2017-10-06:
All is good now for me

yokoyama-flogics, 2022-04-19:
Thank you for the great project, and sorry for commenting on the closed topic.

It's my first evaluation of SkiffOS and I faced a problem very similar to the original question in this post.

I was building SkiffOS 2022.02.1 on Ubuntu 18.04.  In my case the problem seemed to have occurred just because I did not installed the required software.  After running the following command, the problem was resolved.

```bash
$ sudo apt install build-essential
```

This should be careless, but I thought it may be useful for beginner users as much as I am.

Regards,
Atsushi