skiffos/SkiffOS #48: Video card drivers support

frol, 2017-11-26:
I am curious whether there is any interest in video card drivers support, e.g. for Kodi. I have been trying to get a kernel that would support video card drivers and Docker on my Odroid U3. The official kernel has video card drivers support, but it is too old for Docker. The mainline kernel is capable of running Docker, but the proprietary video card drivers are incompatible with it. There is a [mainline kernel fork which brings proprietary drivers from the official kernel](https://forum.odroid.com/viewtopic.php?f=81&t=9342), but it dies unexpectedly in a minute on my box.

P.S. This is a great project! Thank you for working on it!

paralin, 2017-12-19:
Hi @frol welcome to the project!

I would be interested in this. I'm thinking of adding a supplementary package for the odroids that pulls in the relevant video drivers as well.

frol, 2018-02-25:
@paralin Nope... I left my Odroid with the official 3.8 kernel since Docker on ARM is not that critical for me at this point.

paralin, 2018-02-25:
@frol What is your use case? You can disable Docker in Skiff quite easily (BR2_PACKAGE_DOCKER_ENGINE=n) and add Rocket or some other kind of containerization if you want, and/or compile your code into the target.

frol, 2018-02-25:
My use-case is to use Kodi (requires video card support which only works with the Hardkernel proprietary kernel), some small web apps ("smart home"), file sharing and some other random stuff. I don't have any special equipment to debug the kernel issues, so I just give the forked mainline kernel releases a try once in a while, but they have never worked for me.

paralin, 2018-02-25:
@frol We can easily make a Skiff package that just compiles Kodi into the OS and doesn't bother with Docker at all.

paralin, 2018-02-26:
@frol Have a look at the "next" branch - there is a new package named `apps/kodi`. Kodi works fine, when I run it from the CLI as "kodi".

This is just preliminary, the package needs to also add configurations to set the paths to /mnt/persist and whatnot, as well as add a dependency on the persist partition being mounted. But, it's a decent start.

frol, 2018-02-26:
The problem is not with the Kodi itself, but the kernel video driver support. Mainline Linux kernel doesn't have support for the Mali video card Odroid U3 has, and while the forked kernel should bring the support from the 3.8 kernel, the whole system is not stable on that kernel.

paralin, 2018-02-26:
You already explained this. It's not an issue.

closed 2020-05-18 by paralin