skiffos/SkiffOS #51: Skiff odroid ux4 compile error

adriengibrat, 2018-02-03:
Hello,

I want to test SkiffOs on my odroid HC1 (compatible with odroid UX4), I'v ran
```
make # nice help
SKIFF_CONFIG=odroid/xu4 make configure # no issue
make # odroid/xu4 is well selected and configured
make compile # :/ syntax error
```
but i got error when I try to compile. I'm not very familliar with compile log but I think the relevant error is:

> Making all in examples
> /bin/bash ../../libtool --mode=execute ../../dbus/dbus-binding-tool --prefix=some_object --mode=glib-server --output=example-service-glue.h ./example-service.xml
> /bin/bash ../../libtool --mode=execute ../../dbus/dbus-binding-tool --prefix=test_object --mode=glib-server --output=example-signal-emitter-glue.h ./example-signal-emitter.xml
>   File "  File "/home/adrien/Téléchargements/SkiffOS/workspaces/default/host/bin/glib-genmarshal", line 1/home/adrien/Téléchargements/SkiffOS/workspaces/default/host/bin/glib-genmarshal", line 
> SyntaxError1: 
> Non-ASCII character '\xc3' in file /home/adrien/Téléchargements/SkiffOS/workspaces/default/host/bin/glib-genmarshal on line 1, but no encoding declared; see http://python.org/dev/peps/pep-0263/ for details
> SyntaxError: Non-ASCII character '\xc3' in file /home/adrien/Téléchargements/SkiffOS/workspaces/default/host/bin/glib-genmarshal on line 1, but no encoding declared; see http://python.org/dev/peps/pep-0263/ for details
> /usr/bin/make  all-recursive

More complete log file https://gist.github.com/adriengibrat/706f443b8d463da3fa271a993d3793ea

Thanks for creating SkiffOs, sounds very nice and I've found the README very appealing and informative .)

N.B. I was wondering If it could be possible to add a release with attached compiled images using the Github realease API ?

paralin, 2018-02-14:
Hey! Thanks for testing out Skiff. let's get this fixed.

Can you please check out the next branch and test?

```
git fetch origin
git checkout next
git reset --hard origin/next
git clean -xfd # will clean your files
SKIFF_CONFIG=odroid/xu4 make configure compile
```

Also - yes, I am currently working on a distributed build infrastructure right now which I will be able to use to deliver binary builds of Skiff in the future.

adriengibrat, 2018-02-14:
Thanks a lot ;)
Compiling right now, will update ASAP.

adriengibrat, 2018-02-14:
Seems I got the same error 
> SyntaxError: Non-ASCII character '\xc3' in file /home/adrien/Téléchargements/SkiffOS/workspaces/default/host/bin/glib-genmarshal on line 1, but no encoding declared; see http://python.org/dev/peps/pep-0263/ for details
  File "/home/adrien/Téléchargements/SkiffOS/workspaces/default/host/bin/glib-genmarshal", line 1

More logs https://gist.github.com/adriengibrat/706f443b8d463da3fa271a993d3793ea#file-skiff-next-log

adriengibrat, 2018-02-14:
I'll try to compile on a more up to date system (debian 9), my current machine is on a old Linux Mint 17. Could it be the reason of my issues?

adriengibrat, 2018-02-14:
I've try to compile on my odroid (armbian 9), but seems gcc cannot cross-compile there: https://gist.github.com/adriengibrat/706f443b8d463da3fa271a993d3793ea#file-skiff-armbian9-next-log

paralin, 2018-02-14:
Try in a docker container (see the docker dir) if possible. 

The first error you report is not in Skiff, but glib. Not much we can do there.

paralin, 2018-02-14:
See `docker-build` dir or try on the latest Ubuntu on the next branch. 

Also, please make sure you have the latest from the "next" branch - it's been updated a bit.

paralin, 2018-02-14:
@adriengibrat I noticed in your logs you have `myCloudDrive`. Please make sure that when you try running on Armbian on the Odroid (not recommended, due to the superior speed of cross-compiling, but should work) - ensure that the workspace has been cleaned first. You can do this by deleting the directory under `workspaces`.

adriengibrat, 2018-02-14:
Thanks for all the tips, I'll try to build with docker tomorrow ;)

adriengibrat, 2018-02-15:
Compiling in docker right now, seems to work great! 

N.B. ./docker-build deserve a link in the main README ;)

adriengibrat, 2018-02-15:
I got some locale UTF8 warning and an error while compiling systemd 237 https://gist.github.com/adriengibrat/706f443b8d463da3fa271a993d3793ea#file-skiff-docker-next-log
There is no locale set for the `buildroot` user
```
$ locale
LANG=
LANGUAGE=
LC_CTYPE="POSIX"
LC_NUMERIC="POSIX"
LC_TIME="POSIX"
LC_COLLATE="POSIX"
LC_MONETARY="POSIX"
LC_MESSAGES="POSIX"
LC_PAPER="POSIX"
LC_NAME="POSIX"
LC_ADDRESS="POSIX"
LC_TELEPHONE="POSIX"
LC_MEASUREMENT="POSIX"
LC_IDENTIFICATION="POSIX"
LC_ALL=
```
I can't find an easy way to fix it from within the docker container (locked as `buildroot` user without sudo priviledge...).

paralin, 2018-02-16:
Easy to fix with a couple ENV statements in the dockerfile. 

Will commit a fix shortly.

paralin, 2018-02-16:
@adriengibrat Fixed on master and next. Please have a try.

adriengibrat, 2018-02-18:
I got exactly the same issue, with the new Dockerfile (deleted the repo, rebuilt the image, even renamed it) the `buildroot` user does not have any setted locale and thr compile fails at the same step.

adriengibrat, 2018-02-18:
I'm trying to compile in the alpine docker image ...

paralin, 2018-02-18:
@adriengibrat Did you try the `next` branch?

paralin, 2018-02-18:
@adriengibrat The alpine image is a musl based build and musl doesn't really support locales the same way. I can probably fix it, but buildroot also has a lot of stuff that kinda assumes glibc.

I would try to use the basic `Dockerfile` for now.

adriengibrat, 2018-02-18:
yes, using the next branch, with the latest locale related commits.

Alpine has, as you mentionned other issues, i've gave up on alpine docker build!

But i'm back on fixing the locale issue on ubuntu docker build:

I'm compiling right now with the tip about locale explained in https://hub.docker.com/_/ubuntu/
I've added 
```
RUN apt-get install -y locales && localedef -i en_US -c -f UTF-8 -A /usr/share/locale/locale.alias en_US.UTF-8
ENV LANG en_US.utf8
```
in the Dockerfile.
If it works well, I'll submit a small PR

paralin, 2018-02-18:
@adriengibrat https://github.com/paralin/SkiffOS/commit/d2f9109dd9b8519169a5032f8cfc34c5f2e54b1c

This doesn't work?

adriengibrat, 2018-02-18:
[nope](https://github.com/paralin/SkiffOS/issues/51#issuecomment-366526491) got the [same issue](https://github.com/paralin/SkiffOS/issues/51#issuecomment-365900897) and running `locale` always spit empty locale settings.

That's why I've added 
```
RUN localedef -i en_US -c -f UTF-8 -A /usr/share/locale/locale.alias en_US.UTF-8
ENV LANG en_US.utf8
```
Then inside docker when I execute `locale` the output finally show en_US.UTF-8 !
I'm compiling and waiting if I run into any other issue right now;)

paralin, 2018-02-18:
@adriengibrat Awesome, do you want to submit a PR with that, or should I just commit it?

adriengibrat, 2018-02-18:
I'd prefer to check that actually works ;)

paralin, 2018-02-18:
@adriengibrat After your build passes, of course ;)

adriengibrat, 2018-02-19:
See #57

adriengibrat, 2018-02-19:
I'm also trying to compile with a simpler approach, where I just set the ENV var without generating any locales:
```
FROM ubuntu:latest
RUN apt update && \
    apt install -y build-essential \
    git gcc wget curl musl-dev file \
    perl python rsync bc patch unzip cpio
RUN adduser --gecos "Buildroot" --disabled-login --uid 1000 buildroot && \
    mkdir -p /home/buildroot && chown buildroot:buildroot /home/buildroot
USER buildroot
ENV LANG C.UTF-8
WORKDIR /home/buildroot
```

adriengibrat, 2018-02-19:
Thanks for merging PR #57 ;) 
I've made another PR #58 that also solve the locale settings causing glib compile issue, with a simpler solution.

closed 2018-02-19 by adriengibrat

paralin, 2018-02-19:
@adriengibrat nice one

paralin, 2018-02-23:
@adriengibrat The branches have been shuffled around now.

Try out cadvisor:

```
docker run \
 --volume=/:/rootfs:ro \
 --volume=/var/run:/var/run:rw \
 --volume=/sys:/sys:ro \
 --volume=/var/lib/docker/:/var/lib/docker:ro \
 --publish=8080:8080 \
 --detach=true \
 --name=cadvisor \
 braingamer/cadvisor-arm:latest

```