skiffos/SkiffOS #60: Issue with format & install commands

adriengibrat, 2018-02-20:
Hi,

I had some issues to burn my sd card with my fresh build image, as I want to use SkiffOS on my odroid hc1 (xu4 compatible), i've tried:
```
$ sudo bash
$ export ODROID_SD=/dev/mmcblk0 # I'm using a SD card adapter to burn my micro sd scard
$ make cmd/odroid/xu4/format
```
Actually I've run  `cmd/odroid/common/format` as the script to format is in the common/scripts dir, but the actual issue seems there is a variable with the path `/home/buildroot/skiff/` as default or hard coded:
```
> make cmd/odroid/common/format
make[1]: Entering directory '/home/adrien/SkiffOS/build'
make[2]: Entering directory '/home/adrien/SkiffOS/configs/odroid/common/extensions'
make[2]: /home/buildroot/skiff/configs/odroid/common/scripts/format_sd.sh : command not found
```
Then, I've looked at the [format_sd.sh](https://github.com/paralin/SkiffOS/blob/master/configs/odroid/common/scripts/format_sd.sh) & [install_sd.sh](https://github.com/paralin/SkiffOS/blob/master/configs/odroid/common/scripts/install_sd.sh) shell scripts and decided to run them without `make` (I'm a noob with `make`).

I still had minor issues to figure out:
- in my workspaces/default dir the `output` & `staging` symlinks where broken (because they were linking to `/home/buildroot/skiff/workspaces/default`)
- because i'm using a sd card adapter & reader to access my microSD, the partitions names were not sdx1, sdx2, sdx3 but mmcblkx*p1*, mmcblkx*p2* & mmcblkx*p3*, so I had to tweak the scripts everywhere a [partition number is appended to the sdcard path](https://github.com/paralin/SkiffOS/blob/master/configs/odroid/common/scripts/format_sd.sh#L81)

N.B. In the end I manage to burn the SD card, and boot the odroid ;) 
the command I'v ran, with necessary  variables:
```
$ SKIFF_CURRENT_CONF_DIR=/home/adrien/SkiffOS/configs/odroid/xu4/ BUILDROOT_DIR=/home/adrien/SkiffOS/workspaces/default ODROID_SD=/dev/mmcblk0 ./configs/odroid/common/scripts/format_sd.sh
$ SKIFF_CURRENT_CONF_DIR=/home/adrien/SkiffOS/configs/odroid/common BUILDROOT_DIR=/home/adrien/SkiffOS/workspaces/default SKIFF_CONFIG_PATH=/home/adrien/SkiffOS/configs/odroid/xu4 ODROID_SD=/dev/mmcblk0 ./configs/odroid/common/scripts/install_sd.sh
```

paralin, 2018-02-21:
The commands won't work without make, and you figured out why - the paths change between Docker and not docker. 

The fix for this particular GH issue will be to make the scripts detect this and adjust for it during the format / install scripts phase.

paralin, 2018-02-22:
@adriangibrat lots of work in the last couple of days to stabilize docker on the next branch. Please give the latest version there a spin, we are going to promote it to master soon.

adriengibrat, 2018-02-25:
#64 should fix the sd card adapter & reader issue:
> because i'm using a sd card adapter & reader to access my microSD, the partitions names were not sdx1, sdx2, sdx3 but mmcblkxp1, mmcblkxp2 & mmcblkxp3, so I had to tweak the scripts everywhere a partition number is appended to the sdcard path

I've not looked into the path issue:
> The fix for this particular GH issue will be to make the scripts detect [the paths change between Docker and not docker] and adjust for it during the format / install scripts phase.

AFAICR, i didn't manage to understand where the path where set in the `make` scripts.

BTW, I'm compiling/testing the latest next branch

paralin, 2018-02-25:
It's in scripts/primary_env.sh

adriengibrat, 2018-02-25:
yes, I saw scripts/primary_env.sh, I meant where are set/overriden variable/path (pointing to '/home/buildroot'). I'll give it a second try.

paralin, 2018-02-25:
@adriengibrat Check workspaces/.config_myworkspace

closed 2019-12-03 by paralin