skiffos/SkiffOS #63: Update docker version

adriengibrat, 2018-02-24:
Hi,

There is an anoying bug in docker 18.02 https://github.com/docker/for-linux/issues/211

Would be great to bump  to 18.03 as default version in SkiffOS ;)

Btw, how can I update/install another docker version myself  without rebuilding an new image ?

paralin, 2018-02-25:
We saw this error actually, is 2018.03 out?

paralin, 2018-02-25:
At the moment the default version in master is the Q3 2017 patch with the recent revision RC.

paralin, 2018-02-25:
Regarding your question on how to update, the change is in Buildroot: https://github.com/paralin/buidroot you can see usually a series of 3 commits is required to update runc, containerd, and docker-engine.

closed 2018-02-25 by paralin

adriengibrat, 2018-02-26:
Works great
```
docker -v
Docker version v18.03.0-ce-rc1, build v18.03.0-ce-rc1
# docker run \
>  --volume=/:/rootfs:ro \
>  --volume=/var/run:/var/run:rw \
>  --volume=/sys:/sys:ro \
>  --volume=/var/lib/docker/:/var/lib/docker:ro \
>  --publish=8080:8080 \
>  --detach=true \
>  --name=cadvisor \
>  braingamer/cadvisor-arm:latest
Unable to find image 'braingamer/cadvisor-arm:latest' locally
latest: Pulling from braingamer/cadvisor-arm
441134fbd8aa: Pulling fs layer 
3806fda1ef5e: Pulling fs layer 
3a1e065da263: Pull complete 
04bb80fc6b79: Pull complete 
0d2b07e0e24e: Pull complete 
2d7a4d5604aa: Pull complete 
3256ebff284c: Pull complete 
d584d9f1e06d: Pull complete 
7a90402f1002: Pull complete 
138abbfd539b: Pull complete 
Digest: sha256:e922fbdc9633817d872ffbbef906bf182a97afaf825f0047d5907cca68eae9a9
Status: Downloaded newer image for braingamer/cadvisor-arm:latest
031eeefa790e903b5ecbfff6aa21cc18f2561e21adba9dcc8e8a3da2e9b26253
# docker container stop 031eeefa790e903b5ecbfff6aa21cc18f2561e21adba9dcc8e8a3da2e9b26253
031eeefa790e903b5ecbfff6aa21cc18f2561e21adba9dcc8e8a3da2e9b26253
# docker container start 031eeefa790e903b5ecbfff6aa21cc18f2561e21adba9dcc8e8a3da2e9b26253
031eeefa790e903b5ecbfff6aa21cc18f2561e21adba9dcc8e8a3da2e9b26253
```