skiffos/SkiffOS #65: help: categorize available commands

paralin, 2018-02-25:
Make some sort of pretty categorization (by odroid, pi, etc) in the help menu of the available configurations.

gedw99, 2019-01-24:
hey @paralin prettty cool project

I suggest using Mage. AM sure you know it.

https://github.com/search?utf8=%E2%9C%93&q=mage+language%3AGo&type=Repositories&ref=advsearch&l=Go&l=

It can run on any OS, but also can be compiled to a binary.
then you can wrap everything at Dekstop level, but also at the docker level too.

I have been using it to make it easier for users in teams to get going and have reproducibility.
It also makes CI way easyier because all your commands are now in Mage, so you then just confgiure travis to call them.

paralin, 2019-01-25:
@gedw99 Skiff and Buildroot are written entirely in Bash / Makefile. Introducing Go for the build system is unnecessary and would increase complexity in the setup.

PR #96 (2019-10-13): Print available configurations in order
Closes #65
commit 217dac30 (2019-10-13): scripts: print available configurations in order

Closes #65
commit 1f734d5e (2019-11-11): scripts: print available configurations in order

Closes #65
closed 2019-10-13 by paralin