skiffos/SkiffOS #66: Allow docker compose deployment

adriengibrat, 2018-02-28:
Hello,

I'd like to allow [as an option] to deploy docker container from another host.
I've written [some WIP code to add a simple http socket to docker service](https://github.com/paralin/SkiffOS/commit/c7b7224fd7461ba6f963c972b7661e012ff4619f).

I'm not sure how to add this as a proper option, and I know I should add https support.

What do you think about this feature?

paralin, 2018-03-01:
@adriengibrat  Go for it, create a new config package named `docker/remote` that enables the remote socket drop-in.

I also have Kubernetes working on Skiff. We could create a docker-compose orchestrated stand-up of a 3 node cluster as a demo.

I also plan to break out all of the docker stuff into "docker/docker" or so in order to enable users to completely disable Docker in the target image if they want by adding "-docker/docker" to SKIFF_CONFIG (adding negative selectors).

paralin, 2019-02-27:
Docker has now been split into apps/docker

adriengibrat, 2019-03-02:
Thanks for the update, i'm not working on my small Docker file sharing server anymore, I may start it from scratch in a few month. I'll definitivlely use Skiff again!

closed 2019-08-22 by paralin