skiffos/SkiffOS #68: unable to update buildroot submodule

andrewringler, 2018-05-09:
I am trying to get started from master. I do the following, but get an error trying to pull in the buildroot submodule:

    git clone https://github.com/paralin/SkiffOS.git
    cd SkiffOS
    git submodule update --init --recursive
    Submodule 'buildroot' (https://github.com/paralin/buildroot.git) registered for path 'buildroot'
    Cloning into '/Users/andrew/SkiffOS/buildroot'...
    error: Server does not allow request for unadvertised object 
    4aaada448b852b5530612b90ff7394a61b9953fe
    Fetched in submodule path 'buildroot', but it did not contain 
    4aaada448b852b5530612b90ff7394a61b9953fe. Direct fetching of that commit failed.

The above happens with commit b185cf7 using git 2.14.1

I tried with an older commit, 35ef0ca and I had no issues. Is this how we are expected to be pulling in the buildroot submodule, I couldn't find any mention of this in the readme.

tpanum, 2018-06-17:
Having exactly the same issue.

paralin, 2018-07-14:
Typically the submodule would be fetched automatically, but the commit was removed from my Buildroot repo, so the ref doesn't exist.

My apologies, fixed now.

closed 2018-07-14 by paralin

andrewringler, 2018-07-14:
Thanks. Works!