skiffos/SkiffOS #73: Odroid C2: *** No rule to make target 'defconfig'.  Stop.

odroidc2, 2018-09-03:
Hi,

I ran:

SKIFF_CONFIG=odroid/c2 make configure

and the output ends with:
---
make[2]: Entering directory '/root/SkiffOS/workspaces/default'
make[2]: *** No rule to make target 'defconfig'.  Stop.
make[2]: Leaving directory '/root/SkiffOS/workspaces/default'
Makefile:33: recipe for target 'configure' failed
make[1]: *** [configure] Error 2
make[1]: Leaving directory '/root/SkiffOS/build'
Makefile:2: recipe for target 'configure' failed
make: *** [configure] Error 2
---

and nothing else happens. I'm using Ubuntu in the try mode. 
Also, I feel I should mention I'm very new to Linux and I don't really know what I'm doing, but I did search before making this post.

odroidc2, 2018-09-03:
make clean also fails:

---
root@ubuntu:~/SkiffOS# make clean
make[1]: Entering directory '/root/SkiffOS/build'
Are you sure? This will clean and delete your current workspace. [y/N] y
rm: cannot remove '.config': No such file or directory
make[2]: Entering directory '/root/SkiffOS/workspaces/default'
make[2]: *** No rule to make target 'clean'.  Stop.
make[2]: Leaving directory '/root/SkiffOS/workspaces/default'
Makefile:36: recipe for target 'cleanup-workspace' failed
make[1]: *** [cleanup-workspace] Error 2
make[1]: Leaving directory '/root/SkiffOS/build'
Makefile:2: recipe for target 'clean' failed
make: *** [clean] Error 2
---

odroidc2, 2018-09-03:
I ran:

git fetch origin
git checkout next
git reset --hard origin/next
git clean -xfd # will clean your files
SKIFF_CONFIG=odroid/c2 make configure compile

after reading here:

https://github.com/paralin/SkiffOS/issues/51

and it worked until stating I had ran out of space (it's a usb pen drive in live mode), I'm not sure how to change output directory. 

I then entered the same again but without compile and it completed successfully. I thought I would then be taken to a build screen like in buildroot. Clearly I don't know how to use this program and without step by step instructions/an in depth tutorial, it's probably best I stop trying to use it.

paralin, 2018-09-07:
@odroidc2 Okay, looks like you figured it out. The errors you had can be solved by just `rm -rf ./workspaces/default` to clean the workspace and try again (something must have been corrupted somewhere).

The compilation completed successfully, so you would then install to an SD card and boot.

By a "build screen like in buildroot" do you mean something like menuconfig or xconfig? You can still get this:

```
cd workspaces/default
make xconfig
make menuconfig
```

Your changes will last until you run "make configure" again. If you want to persist your changes you need to add a configuration package (create a directory named configs/my/config), and place your options into `configs/my/config/buildroot/mysettings` file like:

```
BR2_PACKAGE_DUMB_INIT=y
```

As a simple example.

closed 2018-10-30 by paralin