skiffos/SkiffOS #79: [User Experience] Can you supply skiffos_rpi.img and skiffos_x64.img?

diyism, 2019-01-23:
For user convenience, can you supply host images, 
for example skiffos_rpi.img for raspberry pi 
and skiffos_x64.img for amd 64 PC?

I found a barge-rpi.img here: https://github.com/bargees/barge-os/releases/tag/2.2.2-rpi
But there is no host image for amd 64 PC.

paralin, 2019-01-26:
Can do. This is something that really needs an automated infrastructure to compile all of the possible configuration combinations. I can work on this part in about a month.

In the meantime, specifically which Pi images do you want? Pi 3, pi 0, etc?

diyism, 2019-01-28:
I saw this: https://github.com/paralin/SkiffOS/tree/master/configs/intel/x64

Can we make a minimal x64 host image and let users to install their platform's specific hardward drivers on it?

paralin, 2019-01-28:
@diyism Yes. By hardware drivers I assume you mean kernel modules. Typically the approach used by distributions is to just compile in all the common kernel modules and then allow them to be hot-loaded at runtime. Skiff has a separate modules image on the disk and loads them on-demand so it is acceptable to use this approach here.

I'll build that image for you.

diyism, 2019-01-28:
Great, i'll give a try to the x64 host image when it's ready.

paralin, 2019-01-28:
@diyism Only thing that will delay this slightly (maybe 1-2 days) is that I need to define the partition layout and GRUB setup for this, which is slightly different than the setup for u-boot and embedded devices. 

http://git.denx.de/?p=u-boot.git;a=blob;f=doc/README.x86

Also I need to see if u-boot might be applicable to x86 as well, that would be convenient.

diyism, 2019-01-28:
Fine, my current PC system is ubuntu 14.04 on /dev/sda1(while /dev/sda2 is idle, i can clone skiffos into it),
my final wish is to replace ubuntu 14.04 with SkiffOS,
so that i can play ubuntu 14.04, ubuntu 16.04, ubuntu 18.04, golang, rust, etc on SkiffOS and docker.

paralin, 2019-01-28:
That's a cool idea and I've also been working on an idea of running a VM hypervisor in Skiff. You could actually run and switch between all those OS at runtime with either docker or a VM. Skiff is an optimal minimal host runtime

paralin, 2019-02-13:
@diyism Still working on this. I have an iso format output for SKIFF_CONFIG=intel/x64 now, and am compiling an image to attach to the 2019.02-rc1 release tag

paralin, 2019-10-09:
Closing this for now, it's a long term goal but some more infra work needs to go into this first.

closed 2019-10-09 by paralin

diyism, 2021-03-26:
Any progress?

BargeOS provide a 13MB img file for digitalocean (https://gist.github.com/diyism/60aa6ca24df772a4928f1aced65e72ee)
but bargeOS project discontinued.

paralin, 2021-03-26:
@diyism see issue #134 

If you want to run on DigitalOcean you can with `intel/desktop` or `intel/x64` config

diyism, 2021-03-26:
> @diyism see issue #134
> 
> If you want to run on DigitalOcean you can with `intel/desktop` or `intel/x64` config
> 
> I plan to add some additional packages related to that

I see the https://github.com/skiffos/SkiffOS/tree/master/configs/intel/x64/scripts/build_image.sh
I'll try:
```
$ git clone https://github.com/skiffos/SkiffOS.git
$ cd SkiffOS
$ make
$ export SKIFF_CONFIG=intel/x64,skiff/core
$ make configure 
$ make compile
```

diyism, 2021-03-26:
no luck:
xzcat /home/malcolm/WorkSpace/singularity/SkiffOS/buildroot/dl/linux/linux-5.10.26.tar.xz | /home/malcolm/WorkSpace/singularity/SkiffOS/workspaces/default/host/bin/tar --strip-components=1 -C /home/malcolm/WorkSpace/singularity/SkiffOS/workspaces/default/build/linux-headers-5.10.26   -xf -
/home/malcolm/WorkSpace/singularity/SkiffOS/workspaces/default/host/bin/tar: arch/arm64/boot/dts/arm: Directory renamed before its status could be extracted
/home/malcolm/WorkSpace/singularity/SkiffOS/workspaces/default/host/bin/tar: arch/arm64/boot/dts: Directory renamed before its status could be extracted
/home/malcolm/WorkSpace/singularity/SkiffOS/workspaces/default/host/bin/tar: arch/arm64/boot: Directory renamed before its status could be extracted
/home/malcolm/WorkSpace/singularity/SkiffOS/workspaces/default/host/bin/tar: Exiting with failure status due to previous errors
make[3]: *** [package/pkg-generic.mk:175: /home/malcolm/WorkSpace/singularity/SkiffOS/workspaces/default/build/linux-headers-5.10.26/.stamp_extracted] Error 2
make[2]: *** [Makefile:23: _all] Error 2
make[2]: Leaving directory '/home/malcolm/WorkSpace/singularity/SkiffOS/workspaces/default'
make[1]: *** [Makefile:64: compile] Error 2
make[1]: Leaving directory '/home/malcolm/WorkSpace/singularity/SkiffOS/build'
make: *** [Makefile:2: compile] Error 2

paralin, 2021-03-26:
@diyism Your computer has failed running "xzcat" on `linux-5.10.26.tar.xz` - really not a whole lot I can do about that. Something is broken on your end. I would recommend deleting the `home/malcolm/WorkSpace/singularity/SkiffOS/workspaces/default/` dir and trying again.

But there's no way to fix that in Skiff you can see what it's trying to do, just extract a .xz.

paralin, 2021-03-26:
@diyism Judging by this I am wondering if you are running this in a Docker container? If so maybe attempt using a ext4 backed path to build? https://github.com/docker/hub-feedback/issues/727

bcopy, 2023-03-02:
Okay I have my intel/desktop SkiffOS ISO, it boots fine from USB.... now I'd like to install SkiffOS to boot the laptop from the hard drive - what steps could someone recommend to install the ISO on the hard disk ?
Since the ISO ships a rEFInd bootloader, would a simple ``dd`` to overwrite the appropriate device do the trick ? (sorry, new to this, I usually use a live CD and click install)

paralin, 2023-03-02:
@bcopy sure, you can dd to overwrite it, or you can create the partitions with parted and then copy over the data.

The instructions for the steam deck apply also for intel/desktop, if you want to use those; https://github.com/skiffos/SkiffOS/tree/master/configs/valve/deck#overwriting-steamos-on-nvme-drive

The most important part is that the refind_linux.conf file has a partition uuid of the root partition. So you have to be sure to update it to use the new partition uuid you created and/or some other root= that refers to your partition.

This definitely could be easier / more automated / better documented (maybe a install script included with the image). I'll spend some time on improving this UX soon for sure.

Lmk if you need any assistance on this. (But feel free to make a new issue)