skiffos/SkiffOS #81: Feasibility of using without Docker?

thirtythreeforty, 2019-02-26:
Hello!  I really like how SkiffOS is organized and am considering using it for a project.  I have a question: how feasible would it be to build images without docker as packaged software?  I am interested in using the layers management, which is really slick, without bundling Docker -- I have 32MB total storage on the board and I'd like to slim down to the very bare minimum.

I see that the SkiffOS core pulls in Docker, so is it "simply" a matter of writing a new base layer that does not require Docker?  Or is each layer automatically put into its own container (in other words, it's very deeply ingrained into the build process)?  If it's the former, I might request an officially-supported Dockerless-base as a feature.

Thank you!

paralin, 2019-02-27:
Very much doable. You can test this in a hacky way by deleting whatever you don't want in skiff/standard/buildroot/*. Just delete the lines related to Docker.

I have plans to onboard new container runtimes - cri-o, kata containers, and others, so it is about time that the Docker core get split into its own package and be made optional.

paralin, 2019-02-27:
I've separated Docker into apps/docker in preparation for removing it as a dependency from skiff/standard.  Before this can happen there are some init scripts that need to check if Docker is enabled or not, but this is trivial and I'll complete that soon.

thirtythreeforty, 2019-02-27:
That's great to hear. I am definitely going to play with it more. I am not up to speed with it yet but ultimately if you want feedback on this I'll be watching with interest.

paralin, 2019-02-27:
You can also add an additional layer with (in a file called buildroot/docker within the layer):

BR2_PACKAGE_DOCKER_ENGINE=n
BR2_PACKAGE_DOCKER_CONTAINERD=n
BR2_PACKAGE_RUNC=n

thirtythreeforty, 2019-02-27:
Oh cool, I didn't think about doing that. I'd prefer to avoid overriding a bunch of stuff in the long term (Yocto BSPs do this and it's very hard to read) but short term, it'll let me play around with changing the core.

paralin, 2019-02-27:
You can overlay your own packages on top of it, and use Skiff as a submodule. It's described how in the readme. 

If you add a custom config, and add the configuration lines I posted above in a buildroot config file, it will effectively disable Docker. You can also use this to add kernel config changes, additional buildroot packages, adjust the toolchain / optimization settings (you can optimize for size for example), etc etc.

cd workspaces/default
make menuconfig # opens the Buildroot config explorer

paralin, 2019-02-27:
I've added an additional way to override the settings temporarily for local experimentation:

You can now simply place this file to disable docker:

```sh
$ cd SkiffOS
$ mkdir ./overrides/buildroot
$ echo "BR2_PACKAGE_DOCKER_ENGINE=n" > ./overrides/buildroot/docker
$ make configure compile
```

Etc, etc. You can place any Skiff configuration package files in "overrides" and override locally the settings.

thirtythreeforty, 2019-02-27:
Cool cool. My toolchain is compiling now.

Another question: Currently there is no way to override a Skiff-level layer dependency, correct?  I found your [comment here](https://github.com/paralin/SkiffOS/issues/66#issuecomment-369500565) suggesting that that was a planned feature.

paralin, 2019-02-27:
Config negation would be cool and I might be able to get that in within the next couple days.

thirtythreeforty, 2019-02-27:
Would be cool, but don't rush it on my behalf!  I also feel like config negation would be the kind of thing you'd want to use really sparingly.  I can see how it'd be useful, but it could also make it very confusing to figure out what is actually getting compiled. So in this example, instead of the base layer providing Docker, each configuration package that provides a container would depend on `apps/docker` individually.

Does that make sense?  My caution is born of navigating override after override of Freescale's BSPs.

paralin, 2019-02-27:
You can override dependencies as configurations later in the SKIFF_CONFIG comma separated chain take precedence. For example you could easily make a "nodocker" package that disables docker

I do plan to remove the apps/docker dependency from skiff/standard soon.

paralin, 2019-03-02:
@thirtythreeforty This is about done.

There are a couple other things to save space: your target board config should support skiff/moduleimg, and you can additionally (although this is not yet in a config package) tell Skiff to mount a loop filesystem instead of loading the entire root system into RAM. This may be preferable if you don't have much memory.

thirtythreeforty, 2019-03-02:
Sweet. 

moduleimg is cool, although currently I'm mounting squashfs rootfs on UBI (and the kernel knows how to mount, so no initrd), so moduleimg it doesn't buy me anything extra in this situation. I'm in a weird situation where the board has plenty of RAM (128MiB) but is tight on storage, so really just pulling Docker out of the base image should be enough.

paralin, 2019-03-17:
This is done now as of https://github.com/paralin/SkiffOS/commit/0ee893202aa48fb20942c9f93c31c96d7a0a4313

closed 2019-03-17 by paralin

thirtythreeforty, 2019-03-17:
Cool. core still appears to depend on apps/docker. Is that by design?

paralin, 2019-03-17:
Yes. "Skiff core" refers to the feature that runs a docker container for each user. That's not required for an actual system. Skiff standard is applied automatically for that.