skiffos/SkiffOS #88: No proper u-boot image after compiling for Odroid XU4

FredericLeuba, 2019-07-30:
Hello,
First, thanks for the great work.
I'm investigating to use your system, but I encounter the following problem and I could not test to the end.

I did:

- SKIFF_CONFIG=odroid/xu4 make configure
- make br/menuconfig (I added Qt5, OpenCV3, Zbar)
- make compile (worked properly)
- sudo bash
- export ODROID_SD=/dev/sdb
- make cmd/odroid/common/format (fails!)

```
root@qtsbuilder# make cmd/odroid/common/format
make[1] : on entre dans le répertoire « /home/plexant/dev/SkiffOS/build »
make[2] : on entre dans le répertoire « /home/plexant/dev/SkiffOS/configs/odroid/common/extensions »
can't find u-boot image at /home/plexant/dev/SkiffOS/workspaces/default/output/images/u-boot-sunxi-with-spl.bin
```

Why is it looking for a sunxi u-boot image ?

Best regards

paralin, 2019-07-31:
Hello! Welcome to Skiff. 

I will check on this and do a full build + boot test on an XU4 this evening. There were some recent changes to the way uboot is compiled for the odroid that I believe caused the issue.

paralin, 2019-08-01:
Please try master, it looks like you're on an older commit.

paralin, 2019-08-02:
@FredericLeuba I've investigated and discovered a malformed config file under configs/odroid/xu4 caused the wrong u-boot binary file to be copied to the output.

Please checkout the latest master and then: `rm -rf ./workspaces/default/build/uboot-2019.04` to force a re-build of u-boot, followed by `make configure compile` in skiff.

closed 2019-08-04 by paralin

paralin, 2019-08-16:
Hey @FredericLeuba - I rolled back u-boot to the official vendor-provided 2017 blobs, fixing some bugs.