skiffos/SkiffOS #94: resize-persist.sh does not work

tianyuanhao, 2019-10-10:
Add `udev.log-priority=debug` to kernel parameters.

```
# journalctl | grep persist
```

```
resize-persist.sh: Cannot find persist device, skipping check.
mount-all.sh: Persist drive is at /mnt/persist
```

```
systemd-udevd: vda2: LINK 'disk/by-label/persist' /usr/lib/udev/rules.d/60-persistent-storage.rules:99
```

`disk/by-label/persist` is populated too late.

paralin, 2019-10-10:
Good point. Need to add a dependency to the unit I suppose.

tianyuanhao, 2019-10-11:
Add the following, copied from `initramfs-tools`, to the beginning of `resize-persist.sh`. Then it would be fine, but I don't know the details, and I'm not sure if there are any side effects.

```patch
 # set -eo pipefail
 
+/lib/systemd/systemd-udevd --daemon --resolve-names=never
+
+udevadm trigger --type=subsystems --action=add
+udevadm trigger --type=devices --action=add
+udevadm settle || true
+
+udevadm control --exit
+
 PERSIST_DEV=/dev/disk/by-label/persist
```

However, the manually started `systemd-udevd` may **occasionally** not exit as expected.

```
# ps | grep systemd-udevd
/lib/systemd/systemd-udevd --daemon --resolve-names=never
/usr/lib/systemd/systemd-udevd
```

commit d9ca01d5 (2019-10-13): skiff/standard: wait for disk/by-label/persist

Closes #94
Related to 7765c4a
commit d3df0ab1 (2019-11-11): skiff/standard: wait for disk/by-label/persist

Closes #94
Related to 7765c4a
closed 2019-10-13 by paralin