skiffos/SkiffOS #103: pcengines/apu2: new platform

matrob, 2020-01-07:
Add support for PC Engines APU2 platform
https://pcengines.ch/apu2.htm
http://lists.busybox.net/pipermail/buildroot/2019-October/261659.html

paralin, 2020-01-10:
Initial support is ready at [the pcengines-apu2 branch](https://github.com/paralin/SkiffOS/tree/pcengines-apu2).

Please test and report back your results. I'm not sure about the boot-up configuration at the moment. It uses coreboot which I don't have experience with.

There's also very little information online about kernel settings needed for full hardware support on this board. I didn't find a device tree for it.

paralin, 2020-05-18:
Closing this as apu2 support was merged. If it doesn't work, it should probably be removed and pushed to a separate feature branch for further dev. Feel free to make another issue if necessary.

closed 2020-05-18 by paralin