skiffos/SkiffOS #109: System Bootup issue related to partition

Chris-Bee, 2020-08-28:
Can not bootup system with persist sector not found.

This issue only concerns the Odroid platform and is related to the fact that SD cards from some manufacturers are not supported by the odroid. I'm not sure if there is something that can be done on the skiff side to solve this issue, mainly because of this line 
```console
A start job is running for SkiffOS …st partition (1min 38s / no limit)
```

Console output over serial:
```console
[   10.812289] systemd[1]: Found device /dev/ttySAC2.
[  OK  ] Found device /dev/ttySAC2.
[   10.834713] systemd[1]: Found device /dev/ttySAC1.
[  OK  ] Found device /dev/ttySAC1.
[   10.873810] systemd[1]: Reached target Sound Card.
[  OK  ] Reached target [0;1;3[   10.881305] mmcblk1: error -110 transferring data, sector 136, nr 120, cmd response 0x900, card status 0x0
9mSound Card.
[   10.921330] mmc_host mmc1: Bus speed (slot 0) = 50000000Hz (slot req 400000Hz, actual 396825HZ div = 63)
[   10.976051] mmc_host mmc1: Bus speed (slot 0) = 50000000Hz (slot req 50000000Hz, actual 50000000HZ div = 0)
[   11.911138] systemd[1]: Found device /dev/disk/by-label/persist.
[  OK  ] Found device /dev/disk/by-label/persist.
[   11.934953] systemd[1]: Starting SkiffOS resize persist partition...
         Starting SkiffOS resize persist partition...
[  *** ] A start job is running for SkiffOS … persist partition (6s / no limit)
[   17.146496] mmcblk1: error -110 transferring data, sector 97767120, nr 8, cmd response 0x900, card status 0x0
[   17.177240] mmc_host mmc1: Bus speed (slot 0) = 50000000Hz (slot req 400000Hz, a[   ***] A start job is running for SkiffOS … persist partition ([   17.231853] mmc_host mmc1: Bus speed (slot 0) = 50000000Hz (slot req 50000000Hz, actual 50000000HZ div = 0)
[ ***  ] A start job is running for SkiffOS …persist partition (11s / no limit)
[   21.143067] random: crng init done
[ ***  ] A start job is running for SkiffOS …st partition (1min 38s / no limit)
```

paralin, 2020-08-28:
```
[   17.146496] mmcblk1: error -110 transferring data, sector 97767120, nr 8, cmd response 0x900, card status 0x0
[   17.177240] mmc_host mmc1: Bus speed (slot 0) = 50000000Hz (slot req 400000Hz, a[   ***]
[   17.231853] mmc_host mmc1: Bus speed (slot 0) = 50000000Hz (slot req 50000000Hz, actual 50000000HZ div = 0)
```

These are kernel errors, at the lowest level - "mmc" - 

Most important is this one - `error -110 transferring data, sector 97767120, nr 8, cmd response 0x900, card status 0x0`

Unfortunately this is a data transfer error to the SD card. The Skiff-related output is systemd attempting to mount the partition before proceeding to `/mnt/persist` but this is only hanging because the mmc transfer is failing.

I would bet that this is mostly related to the Odroid kernel being outdated / buggy in some way.

paralin, 2020-08-28:
Workaround: use a SD card which is compatible with the Odroid kernel.

Solution: upgrade odroid kernel (test out `next` branch it is currently updated to a slightly newer revision, although I don't think it will fix your problem).

It's also possible that the MMC card you were dealing with was broken in some way (they're quite easy to break). Did you try other cards of the same type?

mascheiber, 2020-08-31:
> It's also possible that the MMC card you were dealing with was broken in some way (they're quite easy to break). Did you try other cards of the same type?

Yes, we had this issue with several cards of the same type, however, since switching to a known compatible one, this issue is not present (yet). 

> Solution: upgrade odroid kernel (test out `next` branch it is currently updated to a slightly newer revision, although I don't think it will fix your problem).

Will give that a try then, in combination with #110 to see if these issues were related to the older kernel.

paralin, 2020-09-01:
@mascheiber It's all on `master` now along with 2 other fixes for odroid - for the "Bus speed" message (related to emmc hotplug) and for the s5p firmware missing. `make cmd/odroid/common/install` will not overwrite the persist partition so it can be used to update Skiff independently from your core containers.

I do know there is a particular feature missing in some SD cards which makes them incompatible with Linux. However this is more of a Linux kernel issue rather than a Skiff specific one. If you have a particular kind of card that isn't working, but you expect that it should, it might be most useful to post on https://forum.odroid.com/ where the odroid support team replies usually right away.

paralin, 2020-09-01:
By the way if you know which cards are definitely not compatible, I'd love to know so I can update the Skiff docs.

mascheiber, 2020-09-02:
Well so far had the issues with the `Intenso MicroSDXC Card, UHS-I, 64 GB `, which I have not found on the incompatible list by odroid. Will report it there as well, since all of these issue are mostly related to this card type.

> `make cmd/odroid/common/install` will not overwrite the persist partition so it can be used to update Skiff independently from your core containers.

Thanks will give this a try now, will let you know if this improves these issues.

paralin, 2020-09-02:
@mascheiber Information added to configs/odroid/common/README.md about known cards.

closed 2020-09-06 by paralin