skiffos/SkiffOS #110: Docker does not respond and logon to container not possible

mascheiber, 2020-08-28:
On the OdroidXU4 after booting, when trying to login to the container with `ssh` there is no response. Logging in to root is possible, but trying to restart the docker container is not. Further, any `systemctl <stop|restart> docker` or `docker` commands do not work.

Just to illustrate, when logged on with root, the `docker stop` command does nothing. Similarly `systemctl restart docker` yields no response and just hangs.

```console
# date
Fri Aug 14 12:13:21 UTC 2020
# docker stop -t 0 core
^C
# date
Fri Aug 14 12:18:46 UTC 2020
```

**Current solution:** I have noticed, that when the last poweroff was performed with `poweroff -h` this issue does not appear. Only when the power source was removed when the system was still running, this issue might (not always) be present.

paralin, 2020-08-28:
When this happens it's usually due to I/O blocking. The I/O calls in the kernel will block the thread that made the I/O call until it completes. Most likely, you are booting from a SD card and the /mnt/persist mount (which Docker is using) is being I/O blocked by failing read/write calls.

If you looked at dmesg while this was going on, probably you are going to see the same mmc errors as in #109 

Does this match what you're seeing?

paralin, 2020-08-30:
@mascheiber You may want to try skiff `2020.05.2-rc1` as it has an upgraded Odroid kernel.

mascheiber, 2020-08-31:
> If you looked at dmesg while this was going on, probably you are going to see the same mmc errors as in #109
> 
> Does this match what you're seeing?

Yes, I do get an `EXT4-fs` error for no space left on directory:

```console
[   23.096986] EXT4-fs warning (device mmcblk1p3): 
ext4_dirent_csum_verify:367: inode #265144: 
comm unattended-upgr: No space for directory leaf 
check sum. Please run e2fsck -D.
[   23.111207] EXT4-fs error (device mmcblk1p3): 
ext4_find_entry:1466: inode #26 5144: 
comm unattended-upgr: 
checksumming directory block 0
```

paralin, 2020-09-01:
@mascheiber Are you out of space? `df -h` 

Perhaps a `e2fsck -D /dev/mmcblk1p3` if not, to check + repair the ext partition, as recommended by the error message?

Perhaps some disk corruption happened?

paralin, 2020-09-01:
If it's indeed some kind of disk corruption that occurs during a unclean shutdown, probably it'll be necessary to add a "e2fsck" step to Skiff just before mount.

paralin, 2020-09-02:
Closing for now - let's talk in the other issue if you notice more disk corruption type issues.

closed 2020-09-02 by paralin