skiffos/SkiffOS #111: Docker container build not named (correctly)

mascheiber, 2020-08-28:
After formatting and installing an image, the system was initially booted up on the corresponding hardware (OdroidXU4). The container was built correctly, however the final steps which renames the cached image to `skiff/core-ubuntu-basic:latest` did not happen. 

So after the build was done, a login to `core` would yield the docker build commands.

<details><summary>ssh to core</summary>
<p>

```console
$ ssh core@<IP>
Container setup in progress:
Step 1/60 : FROM ubuntu:bionic
 ---> 40d364ec676a
[... missing commands]
Step 59/60 : WORKDIR /home/core
 ---> Using cache
 ---> 0827bec1e40f
Step 60/60 : ENTRYPOINT ["/lib/systemd/systemd"]
 ---> Using cache
 ---> 20efb3361c7d
```

</p>
</details>

Further, under `root` when listing the docker images the output is as follows

```console
# docker images
REPOSITORY          TAG                 IMAGE ID            CREATED             SIZE
<none>              <none>              20efb3361c7d        9 minutes ago       8.41GB
ubuntu              bionic              40d364ec676a        2 weeks ago         46.8MB
```

Interestingly, when using the finished image with ID `20efb3361c7d` manually with 
```bash
docker run -ti --privileged=true --rm 20efb3361c7d bash
```
and rebooting the system yields to the core image being created correctly
```console
# docker images
REPOSITORY                TAG                 IMAGE ID            CREATED             SIZE
skiff/core-ubuntu-basic   latest              e01e949a91d8        3 minutes ago       8.36GB
<none>                    <none>              20efb3361c7d        46 minutes ago      8.41GB
ubuntu                    bionic              40d364ec676a        2 weeks ago         46.8MB
```
Also the core container is now running on bootup
```console
# docker ps
CONTAINER ID        IMAGE                            COMMAND                  CREATED              STATUS              PORTS               NAMES
aa70329b54b2        skiff/core-ubuntu-basic:latest   "/lib/systemd/systemd"   About a minute ago   Up About a minute                       core
80a487645d51        20efb3361c7d                     "/lib/systemd/system…"   3 minutes ago        Up 3 minutes                            friendly_mclaren
```

paralin, 2020-08-28:
Skiff core will transmit a command equivalent to `docker build -t skiff/core-ubuntu-basic:latest .`

It'd be the Docker daemon which does the tagging at the final step. I don't understand how the core container would have been created if the image compilation step didn't tag the image, because the command skiff core transmits to create the container is `docker create skiff/core-ubuntu-basic:latest`.

Probably a Docker bug? I also am suspecting filesystem issues, given that you've also reported mmc failures in #109 and probably #110

mascheiber, 2020-08-31:
> Probably a Docker bug? I also am suspecting filesystem issues, given that you've also reported mmc failures in #109 and probably #110

Again, the mmc failure could be responsible for that, However, I had this issue with a completely new and unused card (which does not mean it could have been broken already). Hence I'd rather say it's a docker bug, also this image was working fine without any further issues, after I run the `docker run` command.

paralin, 2020-09-01:
Not sure exactly what's going on here, if you do see the problem again though, it'd be very helpful to see `journalctl -u docker` and `journalctl -u skiff-core`. Make sure `skiff-core` finished completely - sometimes Docker takes a while to finalize the image at the end of the build.

paralin, 2020-09-02:
@mascheiber Have you seen this happen since changing SD cards?

mascheiber, 2020-09-03:
So far I haven't. 

Also of my 10+ cards that I installed skiff to, this happened once, making it rather hard to reproduce right now. It was a new, freshly unpacked card though. 

As I mentioned above, after applying the workaround this issue was gone. However, I just looked at my notes from this issue and saw that while building the core container, the power supply was removed at one stage (around step 20/60). After resupplying the power again, the build did continue but did not finish with the final, renaming step.

I tried to reproduce it by reformatting and installing skiff on the card, which had this issue, but this issue has not reappeared. As you said `skiff-core` was probably not finished. Maybe this finishing step took longer than I expected due to the loss of power while building it

paralin, 2020-09-03:
Okay all of that does make sense. What skiff is doing is calling the regular "docker build" command. If the image wasn't tagged it had to have been Docker that made the mistake. If you do see this again let me know and I'll try to see if I can debug it at all, or make an upstream issue report. 

Closing for now. Thanks!

closed 2020-09-03 by paralin

mascheiber, 2020-09-04:
Yes I will, thanks for all your help on this.
Will try to get more detailed logs, next time I encounter this.