skiffos/SkiffOS #112: Docker service not responding or loading

mascheiber, 2020-08-28:
After boot under buildroot, the docker container cannot be started and the service does not respond.

```console
# docker start core
Cannot connect to the Docker daemon at unix:///var/run/docker.sock. Is the docker daemon running?
Error: failed to start containers: core

# docker ps
Cannot connect to the Docker daemon at unix:///var/run/docker.sock. Is the docker daemon running

# systemctl start docker
Job for docker.service failed because the control process exited with error code.
See "systemctl status docker.service" and "journalctl -xe" for details.
```

When looking into the `systemctl status` the output is the following

```console
# systemctl status docker.service
[[0;1;31m●[[0m docker.service - Docker Application Container Engine
     Loaded: loaded (/usr/lib/systemd/system/docker.service; enabled; vendor preset: enabled)
    Drop-In: /usr/lib/systemd/system/docker.service.d
             └─after-init.conf
             /etc/systemd/system/docker.service.d
             └─execstart.conf
             /usr/lib/systemd/system/docker.service.d
             └─ramdisk.conf
     Active: [[0;1;31mfailed[[0m (Result: exit-code) since Wed 2020-08-05 08:21:57 UTC; 1min 4s ago
TriggeredBy: [[0;1;31m●[[0m docker.socket
       Docs: https://docs.docker.com
    Process: 822 ExecStart=/usr/bin/dockerd -H fd:// --data-root=/mnt/persist///skiff/docker --log-dri
ver=journald [[0;1;31m(code=exited, status=1/FAILURE)[[0m
   Main PID: 822 (code=exited, status=1/FAILURE)

Aug 05 08:21:57 skiff-65c7ed4 systemd[1]: docker.service: Scheduled restart job, restart counter is at
 3.
Aug 05 08:21:57 skiff-65c7ed4 systemd[1]: Stopped Docker Application Container Engine.
Aug 05 08:21:57 skiff-65c7ed4 systemd[1]: [[0;1;38;5;185m[[0;1;39m[[0;1;38;5;185mdocker.service: Start
 request repeated too quickly.[[0m
Aug 05 08:21:57 skiff-65c7ed4 systemd[1]: [[0;1;38;5;185m[[0;1;39m[[0;1;38;5;185mdocker.service: Faile
d with result 'exit-code'.[[0m
Aug 05 08:21:57 skiff-65c7ed4 systemd[1]: [[0;1;31m[[0;1;39m[[0;1;31mFailed to start Docker Applicatio
n Container Engine.[[0m
```

Further stopping and starting the service does not help

```console
# systemctl stop docker
# systemctl start docker
Job for docker.service failed because the control process exited with error code.
See "systemctl status docker.service" and "journalctl -xe" for details.
```

paralin, 2020-08-28:
@mascheiber I hate to say it but all of your issues are probably caused by the mmc failures.

paralin, 2020-08-30:
By the way - if you see this again, the best way to get logs for the Docker service to debug it would be `journalctl -u docker` - please attach. Thanks!

mascheiber, 2020-08-31:
> @mascheiber I hate to say it but all of your issues are probably caused by the mmc failures.

Yes, this could very well be. However, this would mean we had many failed mmcs in combination with the odroid.

> By the way - if you see this again, the best way to get logs for the Docker service to debug it would be `journalctl -u docker` - please attach. Thanks!

Sadly do not have the card with the image causing this issue anymore, but will do when I see this again. 

Again, I do agree that all these issues might have been caused by the faulty mmcs, we did want to make sure that this is the most probable cause of our problems. Again, will try the new kernel as mentioned in #110 and this card type. Otherwise, we have already switched to a different card type, where these issues have not been present yet.

paralin, 2020-09-01:
@mascheiber Sounds good. I have not encountered any of these while testing with 2 HC2 boards, 2 XU4 boards, even after sometimes months of uptime. I'm using this type of SD card - maybe it's just better compatible - `SanDisk 64GB Ultra microSDXC UHS-I Memory Card`

mascheiber, 2020-09-02:
Yes we have switched to the same SD card manufacturer (`SanDisk 64GB Extreme microSDXC UHS-I U3 A2`), which is working way better on our side as well. The previous one (`Intenso MicroSDXC Card, UHS-I, 64 GB`) caused these issues mostly.

paralin, 2020-09-02:
Sounds good, closing this for now.

closed 2020-09-02 by paralin