skiffos/SkiffOS #113: Issue with no serial config

Chris-Bee, 2020-08-28:
The boot.ini has to be changed manually after the SD card is formatted and installed. This only applies if the "no_serial" config is used. The PI does still boot but the serial port is not available. The odroid never boots if the line is not changed.

(Odroid) This line in the boot.ini has to be changed. Setting $condev to an empty string is not sufficient.
```diff
  #---------------------------------
  # Boots from memory.
  # --------------------------------
- setenv bootmem "console=tty1 ${condev} root=/dev/initrd ro ramdisk_size=100000 fsck.repair=yes net.ifnames=0 no_console_suspend"
+ setenv bootmem "root=/dev/initrd ro ramdisk_size=100000 fsck.repair=yes net.ifnames=0 no_console_suspend"
```

I will provide the line that needs to be changed on the PI soon.

paralin, 2020-08-28:
The `noserial` package creates a file which the "install_sd" script will detect & patch the startup scripts. 

Probably a better solution is to have configuration fragments which can be added to the images dir and copied into both rpi and odroid and get loaded in parallel with config.txt

The configuration changes would then be specified in resources/rpi/config.txt.d/my-config.txt (can specify an option to disable serial there, and a cmdline file as well), and resources/odroid/boot.ini.d/my-boot.ini (or so).

Alternatively: it's possible to run u-boot on the Raspberry Pi boards, I am looking at enabling that so that we can have u-boot across all Skiff boards and standardize the boot.ini.d drop-in format.

paralin, 2020-09-10:
For rpi - u-boot support is coming soon, and the drop in config dir as mentioned above.

For odroid - this should be fixed as the condev now includes both console= options.

closed 2020-09-10 by paralin