skiffos/SkiffOS #119: Compilation error

mascheiber, 2020-09-25:
### Summary
I recently exchanged my laptops and started to recompile our odroidXU4 image using the latest skiff_os version, using a freshly setup Ubuntu 20.04. However i ran into an error with the sha256 has of the `linux-firmware-20200817.tar.gz` while compiling, and subsequent could not finish the compilation. Tried it already several times, at different times of day, just to be sure that this is not network related or something similar.

Here is the full output when this error appears

```console
[ommited output]
>>> linux-firmware 20200817 Downloading                                                                                          
Reinitialized existing Git repository in /data/WS/docker_ws/cns_skiffos/skiffos/buildroot/dl/linux-firmware/git/.git/            
Fetching all references                                                                                                          
warning: redirecting to https://git.kernel.org/pub/scm/linux/kernel/git/firmware/linux-firmware.git/                             
warning: redirecting to https://git.kernel.org/pub/scm/linux/kernel/git/firmware/linux-firmware.git/                             
Could not fetch special ref '20200817'; assuming it is not special.                                                              
ERROR: linux-firmware-20200817.tar.gz has wrong sha256 hash:                                                                     
ERROR: expected: 23567c0d75734167a8442419299cd7d3ab9a27582b20d4a6feeda2b1da68ed2b                                                
ERROR: got     : dedda2f97b1925696cb259d1587efa506b0d909e3a52d68f28c5e557d1ff039e                                                
ERROR: Incomplete download, or man-in-the-middle (MITM) attack                                                                   
--2020-09-25 12:23:10--  http://sources.buildroot.net/linux-firmware/linux-firmware-20200817.tar.gz                              
Resolving sources.buildroot.net (sources.buildroot.net)... 104.26.0.37, 172.67.72.56, 104.26.1.37, ...                           
Connecting to sources.buildroot.net (sources.buildroot.net)|104.26.0.37|:80... connected.                                        
HTTP request sent, awaiting response... 404 Not Found                                                                            
2020-09-25 12:23:10 ERROR 404: Not Found.                                                                                        
                                                                                                                                 
--2020-09-25 12:23:10--  http://sources.buildroot.net/linux-firmware-20200817.tar.gz                                             
Resolving sources.buildroot.net (sources.buildroot.net)... 104.26.1.37, 172.67.72.56, 104.26.0.37, ...                           
Connecting to sources.buildroot.net (sources.buildroot.net)|104.26.1.37|:80... connected.                                        
HTTP request sent, awaiting response... 404 Not Found                                                                            
2020-09-25 12:23:10 ERROR 404: Not Found.                                                                                        

make[3]: *** [package/pkg-generic.mk:171: /data/WS/docker_ws/cns_skiffos/skiffos/workspaces/hummy_xu4/build/linux-firmware-202008
17/.stamp_downloaded] Error 1
make[2]: *** [Makefile:23: _all] Error 2
make[2]: Leaving directory '/data/WS/docker_ws/cns_skiffos/skiffos/workspaces/hummy_xu4'                                         
make[1]: *** [Makefile:64: compile] Error 2
make[1]: Leaving directory '/data/WS/docker_ws/cns_skiffos/skiffos/build'                                                        
make: *** [Makefile:2: compile] Error 2
```

#### Steps to reproduce
1. On previous `master` branch (dbb9450) or commit 65c7ed4f (both result in the same error)
2. Source our personal environments 
3. Goto `skiff_os` and select the desired configuration, `env/hummy` is the personal environment
```bash
cd skiff_os
export SKIFF_WORKSPACE=hummy_xu4
export SKIFF_CONFIG=odroid/xu4,env/hummy,skiff/noserial
```
4. Configure and compile with `make configure compile`

Please let me know if you need any further information or if there are any compile logfiles I can provide you to debug this

paralin, 2020-09-25:
Something is weird here, because on my end, with a fresh download, the hash of that tag on the linux-firmware repo is correct:

```
From http://git.kernel.org/pub/scm/linux/kernel/git/firmware/linux-firmware
 * [new branch]      main       -> origin/main
 * [new branch]      master     -> origin/master
 * [new tag]         20200918   -> 20200918
 * [new tag]         20190312   -> 20190312
 * [new tag]         20190416   -> 20190416
 * [new tag]         20190514   -> 20190514
 * [new tag]         20190618   -> 20190618
 * [new tag]         20190717   -> 20190717
 * [new tag]         20190815   -> 20190815
 * [new tag]         20190923   -> 20190923
 * [new tag]         20191022   -> 20191022
 * [new tag]         20191215   -> 20191215
 * [new tag]         20200122   -> 20200122
 * [new tag]         20200316   -> 20200316
 * [new tag]         20200421   -> 20200421
 * [new tag]         20200519   -> 20200519
 * [new tag]         20200619   -> 20200619
 * [new tag]         20200721   -> 20200721
 * [new tag]         20200817   -> 20200817
warning: redirecting to https://git.kernel.org/pub/scm/linux/kernel/git/firmware/linux-firmware.git/
Could not fetch special ref '20200817'; assuming it is not special.
linux-firmware-20200817.tar.gz: OK (sha256: 23567c0d75734167a8442419299cd7d3ab9a27582b20d4a6feeda2b1da68ed2b)
```

For some reason the download on your end has incorrect integrity, and is coming across as hash dedda2f97b1925696cb259d1587efa506b0d909e3a52d68f28c5e557d1ff039e.

paralin, 2020-09-25:
OK @mascheiber I've updated all branches to use a direct HTTP download instead of Git. This should fix the hash problem.

Closing for now, please comment if this is still not fixed.

closed 2020-09-25 by paralin

mascheiber, 2020-10-07:
Hi, sorry for answereing this late but with these changes and on the newest master, the compile worked smoothly again!

Thanks a lot for your help!

paralin, 2020-10-08:
@mascheiber Glad to hear, let me know if you need any other help.