skiffos/SkiffOS #134: Release build images?

eldondev, 2021-02-01:
The idea of skiffos looks great! But one of the benefits of reproducible builds is being able to publish your build and verify it against somebody else's build. Is there a set of builds that you plan on publishing? Did I just somehow miss the download link :sleepy: ? Specifically being able to launch a qemu build directly from an iso/disk image would be great. Thanks!

paralin, 2021-02-01:
Hey, thanks for the report. Yes, this is important. I was actually looking yesterday at refactoring so that intel/desktop produces an ISO that works directly in Qemu or on a machine. I use the apple/macbook install script and rEFInd to do this right now. While I did manage to add a package to build it from source, it hasn't been updated in forever, and apparently people are just using grub directly now anyway. So the next step is to ensure the grub configuration works on all targets.

I'll push the commit to `next` with the iso refactor shortly and will also post a link to an image here produced from a GitHub build artifact.

paralin, 2021-02-02:
By the way, if you want to build+run in qemu, this is how:

```
export SKIFF_WORKSPACE=qemu
export SKIFF_CONFIG=virt/qemu,skff/core
make compile # build the system
make cmd/virt/qemu/run # launch qemu
```

paralin, 2021-02-06:
@eldondev The GitHub actions currently produce a `virt/qemu,core/gentoo` configuration, I'm working on a system which will auto-build a variety of configs to attach to each release. Also, something like this:

```
export SKIFF_WORKSPACE=qemu
export SKIFF_CONFIG=virt/qemu,core/gentoo
make fetch-prebuilt
# skiff auto-downloads pre-built image for this config to images/ dir
make cmd/virt/qemu/run # all install commands work as usual
```

diyism, 2021-02-18:
Great,  wish an ISO or dd image release works on an x64 machine.

paralin, 2021-02-18:
I'm working on producing ISOs and the scripts to produce them for the next 2020.11.7 release, which will be made by the GitHub actions.