skiffos/SkiffOS #140: SD Card will not format correctly for pinephone

EnderNightLord-ChromeBook, 2021-03-07:
`root@camden-elitebook:/home/camden/SkiffOS# make cmd/pine64/common/format
make[1]: Entering directory '/home/camden/SkiffOS/build'
make[2]: Entering directory '/home/camden/SkiffOS/configs/pine64/common/extensions'
Are you sure? This will completely destroy all data. [y/N] y
Verify that '/dev/sdb' is the correct device. Be sure. [y/N] y
+ set -e
+ echo 'Formatting device...'
Formatting device...
+ sudo dd if=/dev/zero of=/dev/sdb bs=8k count=13 oflag=dsync
13+0 records in
13+0 records out
106496 bytes (106 kB, 104 KiB) copied, 0.0222609 s, 4.8 MB/s
+ sudo parted /dev/sdb mklabel msdos
Information: You may need to update /etc/fstab.

+ echo 'Making boot partition...'                                         
Making boot partition...
+ sudo parted -a optimal /dev/sdb mkpart primary fat32 128MiB 510MiB
Information: You may need to update /etc/fstab.

+ sudo parted /dev/sdb set 1 boot on                                      
Information: You may need to update /etc/fstab.

+ sudo parted /dev/sdb set 1 lba on                                       
Information: You may need to update /etc/fstab.

+ PINE64_SD_SFX=/dev/sdb                                                  
+ '[' -b /dev/sdbp1 ']'
+ mkfs.vfat -F 32 /dev/sdb1
mkfs.fat 4.1 (2017-01-24)
mkfs.vfat: unable to open /dev/sdb1: No such file or directory
make[2]: *** [Makefile:2: format] Error 1
make[2]: Leaving directory '/home/camden/SkiffOS/configs/pine64/common/extensions'
make[1]: *** [Makefile:10: cmd/pine64/common/format] Error 2
make[1]: Leaving directory '/home/camden/SkiffOS/build'
make: *** [Makefile:2: cmd/pine64/common/format] Error 2
root@camden-elitebook:/home/camden/SkiffOS# 
`

paralin, 2021-03-07:
Hi, thanks for the report. I'm actually looking into revamping the Pinephone boot / format at the moment. 

Your OS is not recognizing the updated partition layout in time. `partprobe` and some sleeps probably need to be added to the script. Do the partition files appear in /dev after a little while?

Thanks

paralin, 2021-03-07:
@EnderNightLord-ChromeBook I committed a fix to `master` branch which creates the partitions first, then calls `partprobe`, then waits, then formats the partitions. Does that fix the issue?

EnderNightLord-ChromeBook, 2021-03-08:
Hey man, thanks, I'll give it a test later today.

paralin, 2021-03-08:
@EnderNightLord-ChromeBook I really like what you've done with dahliaos and will be reading into it a lot more. Perhaps SkiffOS could help fill in the "buildroot" section by providing the device support packages. You can add it as a sub-module [similar to this](https://github.com/skiffos/skiff-ext-example) and add custom/modular config layers which bring in the rest of what you need.

EnderNightLord-ChromeBook, 2021-03-08:
I've been working on Fuchsia and Dahlia Linux to run on the pinephone ;) 

We used to have this mobile flutter test UI, I need to get a look at it again sometime

EnderNightLord-ChromeBook, 2021-03-08:
I have FuchsiaOS shim and stuff but the devs changed the build file format and stuff so I got confused, same with my friend who is a google dev, he took a while to get used to it lol.

EnderNightLord-ChromeBook, 2021-03-13:
No problem

EnderNightLord-ChromeBook, 2021-03-13:
It still doesn't want to work it seems

```
+ echo 'Waiting for partprobe...'                                         
Waiting for partprobe...
+ partprobe /dev/sdb
+ sleep 2
+ partprobe /dev/sdb
+ PINE64_SD_SFX=/dev/sdb
+ '[' -b /dev/sdbp1 ']'
+ '[' '!' -b /dev/sdb1 ']'
+ echo 'Warning: it appears your kernel has not created partition files at /dev/sdb.'
Warning: it appears your kernel has not created partition files at /dev/sdb.
+ echo 'Formatting boot partition...'
Formatting boot partition...
+ mkfs.vfat -F 32 /dev/sdb1
mkfs.fat 4.1 (2017-01-24)
mkfs.vfat: unable to open /dev/sdb1: No such file or directory
make[2]: *** [Makefile:2: format] Error 1
make[2]: Leaving directory '/home/camden/SkiffOS/configs/pine64/common/extensions'
make[1]: *** [Makefile:10: cmd/pine64/common/format] Error 2
make[1]: Leaving directory '/home/camden/SkiffOS/build'
make: *** [Makefile:2: cmd/pine64/common/format] Error 2
root@camden-elitebook:/home/camden/SkiffOS#
```

paralin, 2021-03-13:
After running parted on the disk, the partition block files are not being created by your kernel. It should make partition files at sdb1, sdb2, sdb3. What OS are you running this on?