skiffos/SkiffOS #142: ci: build issue when host system is arm64

paralin, 2021-03-24:
When building on an arm64 system, we encounter the following error building `host-python`:

```
/build/host-python-2.7.18/Modules/_ctypes/libffi/src/aarch64/sysv.S: Assembler messages:
/build/host-python-2.7.18/Modules/_ctypes/libffi/src/aarch64/sysv.S:145: Error: constant expression required at operand 2 -- `tbz x23,#AARCH64_FFI_WITH_V_BIT,1f'
/build/host-python-2.7.18/Modules/_ctypes/libffi/src/aarch64/sysv.S:174: Error: constant expression required at operand 2 -- `tbz x23,#AARCH64_FFI_WITH_V_BIT,1f'
/build/host-python-2.7.18/Modules/_ctypes/libffi/src/aarch64/sysv.S:282: Error: constant expression required at operand 2 -- `tbz x0,#AARCH64_FFI_WITH_V_BIT,1f'
/build/host-python-2.7.18/Modules/_ctypes/libffi/src/aarch64/sysv.S:301: Error: constant expression required at operand 2 -- `tbz x0,#AARCH64_FFI_WITH_V_BIT,1f'
/build/host-python-2.7.18/Modules/_ctypes/libffi/src/aarch64/sysv.S:260: Error: undefined symbol AARCH64_CALL_CONTEXT_SIZE used as an immediate value
```

Example of a CI failure: https://github.com/skiffos/SkiffOS/runs/2172698881

Issue filed with Buildroot: https://bugs.busybox.net/show_bug.cgi?id=13661

paralin, 2021-06-17:
@ootoovak did some testing of this: https://github.com/skiffos/SkiffOS/issues/155

Still is an issue with Ubuntu build system

paralin, 2021-06-21:
Confirmed fixed & merged

closed 2021-06-21 by paralin

ootoovak, 2021-06-22:
Thanks for following this up @paralin! I'll try it again this week. 😃