skiffos/SkiffOS #143: Incorrect selection of kernel headers: expected 5.9.x, got 5.11.x

ootoovak, 2021-04-07:
Using Docker on an Intel Mac to try to build SkiffOS I am getting the following error:
```
>>> linux-headers custom Installing to staging directory
(cd /home/buildroot/skiff/workspaces/default/build/linux-headers-custom; PATH="/home/buildroot/skiff/workspaces/default/host/bin:/home/buildroot/skiff/workspaces/default/host/sbin:/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin" /usr/bin/make -j5 ARCH=arm64 HOSTCC="/home/buildroot/skiff/workspaces/default/host/bin/ccache /usr/bin/gcc" HOSTCFLAGS="" HOSTCXX="/home/buildroot/skiff/workspaces/default/host/bin/ccache /usr/bin/g++" INSTALL_HDR_PATH=/home/buildroot/skiff/workspaces/default/host/aarch64-buildroot-linux-gnu/sysroot/usr headers_install)
  INSTALL /home/buildroot/skiff/workspaces/default/host/aarch64-buildroot-linux-gnu/sysroot/usr/include
if ! support/scripts/check-kernel-headers.sh  /home/buildroot/skiff/workspaces/default/build  /home/buildroot/skiff/workspaces/default/host/aarch64-buildroot-linux-gnu/sysroot  5.9 strict; then exit 1; fi
Incorrect selection of kernel headers: expected 5.9.x, got 5.11.x
make[3]: *** [package/pkg-generic.mk:290: /home/buildroot/skiff/workspaces/default/build/linux-headers-custom/.stamp_staging_installed] Error 1
make[2]: *** [Makefile:23: _all] Error 2
make[2]: Leaving directory '/home/buildroot/skiff/workspaces/default'
make[1]: *** [Makefile:64: compile] Error 2
make[1]: Leaving directory '/home/buildroot/skiff/build'
make: *** [Makefile:2: compile] Error 2
```

As mentioned the build is attempted on an Intel Mac in Docker on branch `2021.02-rc3` using the following commands:
```
export SKIFF_CONFIG=apps/docker,apps/yacht,core/gentoo,pine64/rockpro64,skiff/core
make configure
make compile
```
The error occurs on the compile step.

I'm pretty sure I had a similar build error on the `2020.11.7` branch as well (at least it didn't compile either).

ootoovak, 2021-04-07:
I'm not experienced enough compiling Linuxes to know what to do from here. At first I thought it might be from trying to pull in ayufan's patches that were from a different kernel but looks like that is only built (at newest) to 5.4, so that doesn't seem to fit.

I'm happy to try test out any changes, if there are attempted fixes to this. Thanks!

paralin, 2021-04-07:
Hi, I was able to reproduce and also fix this issue by dropping a old config line in pine64/common. Please try again on latest `master` branch.

ootoovak, 2021-04-07:
I changed:
```
pine64/common,skiff/moduleimg
```
to
```
skiff/moduleimg
```
in the file `configs/pine64/rockpro64/metadata/dependencies`.

Which got me a new error, which is progress! 

```
make[6]: *** [../o-iterator.mk:9: /home/buildroot/skiff/workspaces/default/build/glibc-2.32-37-g760e1d287825fa91d4d5a0cc921340c740d803e2/build/dlfcn/stamp.oS] Error 1
make[5]: *** [Makefile:470: dlfcn/subdir_lib] Error 2
make[4]: *** [Makefile:9: all] Error 2
make[3]: *** [package/pkg-generic.mk:250: /home/buildroot/skiff/workspaces/default/build/glibc-2.32-37-g760e1d287825fa91d4d5a0cc921340c740d803e2/.stamp_built] Error 2
make[2]: *** [Makefile:23: _all] Error 2
make[2]: Leaving directory '/home/buildroot/skiff/workspaces/default'
make[1]: *** [Makefile:64: compile] Error 2
make[1]: Leaving directory '/home/buildroot/skiff/build'
make: *** [Makefile:2: compile] Error 2
```

Perhaps the difference is the selection I used for `SKIFF_CONFIG`?

paralin, 2021-04-07:
You can't just drop the common package and expect it to work. It's needed.

Undo that change, delete your workspace (to clean) i.e. `rm -rf ./workspaces/default` and try again with `SKIFF_CONFIG=pine64/rockpro64,core/gentoo,apps/yacht`.

paralin, 2021-04-07:
@ootoovak Just noticed that I didn't push the commit, oops. It's commit 14282178d7801738 - give it a try with that one applied.

ootoovak, 2021-04-07:
Oh, nice, thank you! I'll try test this out tomorrow and get back to you. I appreciate the fix, and all the work you've done on this project! 😃 ✨

ootoovak, 2021-04-08:
Ok @paralin this is now the error I'm currently getting on the main branch. I'm just reading thought it now to try figure out what is going on. If you have any ideas it would be much appreciated. 😃 
```
a - resource/nice.os
/home/buildroot/skiff/workspaces/default/host/lib/gcc/aarch64-buildroot-linux-gnu/10.2.0/../../../../aarch64-buildroot-linux-gnu/bin/ar: mssgwsgadebug/backtrace.os: No such file or directory
make[6]: *** [../o-iterator.mk:9: /home/buildroot/skiff/workspaces/default/build/glibc-2.32-37-g760e1d287825fa91d4d5a0cc921340c740d803e2/build/libc_nonshared.a] Error 1
make[6]: *** Waiting for unfinished jobs....
sed -e 's@ /home/buildroot/skiff/workspaces/default/build/glibc-2\.32-37-g760e1d287825fa91d4d5a0cc921340c740d803e2/build/@ $(common-objpfx)@g' -e 's@^/home/buildroot/skiff/workspaces/default/build/glibc-2\.32-37-g760e1d287825fa91d4d5a0cc921340c740d803e2/build/@$(common-objpfx)@g' -e 's@  *\.\.\/\([^ 	\]*\)@ $(..)\1@g' -e 's@^\.\.\/\([^ 	\]*\)@$(..)\1@g' \
    /home/buildroot/skiff/workspaces/default/build/glibc-2.32-37-g760e1d287825fa91d4d5a0cc921340c740d803e2/build/dl-link.h.dT > /home/buildroot/skiff/workspaces/default/build/glibc-2.32-37-g760e1d287825fa91d4d5a0cc921340c740d803e2/build/dl-link.h.dT2
rm -f /home/buildroot/skiff/workspaces/default/build/glibc-2.32-37-g760e1d287825fa91d4d5a0cc921340c740d803e2/build/dl-link.h.dT
mv -f /home/buildroot/skiff/workspaces/default/build/glibc-2.32-37-g760e1d287825fa91d4d5a0cc921340c740d803e2/build/dl-link.h.dT2 /home/buildroot/skiff/workspaces/default/build/glibc-2.32-37-g760e1d287825fa91d4d5a0cc921340c740d803e2/build/dl-link.h.d
mv -f /home/buildroot/skiff/workspaces/default/build/glibc-2.32-37-g760e1d287825fa91d4d5a0cc921340c740d803e2/build/dl-link.hT /home/buildroot/skiff/workspaces/default/build/glibc-2.32-37-g760e1d287825fa91d4d5a0cc921340c740d803e2/build/dl-link.h
make[5]: *** [Makefile:470: elf/subdir_lib] Error 2
make[4]: *** [Makefile:9: all] Error 2
make[3]: *** [package/pkg-generic.mk:250: /home/buildroot/skiff/workspaces/default/build/glibc-2.32-37-g760e1d287825fa91d4d5a0cc921340c740d803e2/.stamp_built] Error 2
make[2]: *** [Makefile:23: _all] Error 2
make[2]: Leaving directory '/home/buildroot/skiff/workspaces/default'
make[1]: *** [Makefile:64: compile] Error 2
make[1]: Leaving directory '/home/buildroot/skiff/build'
make: *** [Makefile:2: compile] Error 2
```

paralin, 2021-04-08:
@ootoovak it's a build failure in glibc

Did you fully delete the directory "skiff/workspaces/default/" before attempting the build?

ootoovak, 2021-04-08:
A bit more background on the above:

That was compiled on a completely clean slate btw. Just to be sure I trashed, the docker container, image, and the whole `SkiffOS` folder and re-cloned it (likely overdoing it I know when I could just delete `./workspaces/default` but wanted to be extra sure 😂).

Once doing all that (using the Docker build `./build.sh` and `./run.sh` scripts) in Docker I ran:
```
SKIFF_CONFIG=pine64/rockpro64,core/gentoo,apps/yacht
make configure
make compile
```
Which gave me the above error.

ootoovak, 2021-04-08:
@paralin just saw your comment, see my late response above. I was built on a completely new everything.

paralin, 2021-04-11:
@ootoovak I did the exact same steps, and the build succeeded. Could you try deleting workspaces/default and workspaces/.config_default as well as `docker rm -f skiff-build` and try again with latest `master`? I also pinned the Ubuntu version to `focal` on master.

If it still doesn't work, it's probably because the backing filesystem is different - on MacOS the backing fs will be apple journaled filesystem, which is possibly breaking the glibc build. Let's give it a try and see.

If it's not working still - you could try (inside the docker container) - `rm -rf ./workspaces; mkdir /build; ln -s /build ./workspaces` and then build again to use the Docker filesystem instead of the mounted one.

I will upload the results of the build in case you want to use the pre-built version for now until we figure out why the MacOS container is not working.

paralin, 2021-04-11:
Here is the fully compiled image as a ISO: [RockPro64 ISO](https://drive.google.com/file/d/1HoSF-T1guewZqc3K6BblObDk6QIuZvbP/view?usp=sharing)

You can copy it to your card with something like Balena Etcher or `gzip -d ./skiffos-rockpro64.img.gz` and  `dd if=skiffos-rockpro64.img of=/dev/mysdcardpathhere`

Note: the rockpro64 ethernet appears to have an issue right now, to add wifi config, follow these steps:

 1. Flash the image to your SD card
 2. Mount the 3rd partition (named persist) (this is ext4, not sure if macos can handle?)
 3. Create the directory "skiff/keys" and put your public key at "skiff/keys/id_rsa.pub"
 4. Create the directory "skiff/connections" and add text file "skiff/connections/mywifi"

```
[connection]
id=MyWifi
uuid=91291558-93f8-4bae-aaa8-a1c528d32bbc
type=wifi

[wifi]
mode=infrastructure
ssid=MyWifi

[wifi-security]
auth-alg=open
key-mgmt=wpa-psk
psk=mywifipasswordhere

[ipv4]
method=auto
ignore-auto-dns=true
dns=8.8.8.8;8.8.4.4

[ipv6]
addr-gen-mode=stable-privacy
method=auto
```

Put the SD card into the rockpro64 and press (& hold) the power button to start it. It should boot and connect to the wifi, and then be accessible with SSH at "root" user with your key.

ootoovak, 2021-04-11:
Thank you for all that, I am going to try the steps again to confirm, both on the macOS directly, as well as Ubuntu running in VirtualBox to see if I can get it to compile on Linux but not on Mac with the current main branch. I'll aim to report back what I find. Thanks for all the info as well as sharing the pre-built image!

ootoovak, 2021-04-13:
@paralin confirmed the Docker build works on Ubuntu but not on Mac at the moment. I haven't had a chance to do like you say and delete the workspace folder from within the Docker instance so the folder structure is made by a Linux but I'll try give that a go at some point. Just takes so long to compile on my machine! 😅

paralin, 2021-04-13:
@ootoovak Going to close this particular issue for now, please see https://github.com/skiffos/SkiffOS/issues/146 about the macos build & post there w/ any issues. Thanks again for testing.

closed 2021-04-13 by paralin