skiffos/SkiffOS #145: Format error in `make cmd/pi/common/format`

RickCarlino, 2021-04-09:
# Steps to Reproduce

 * I started with a 15 GB memory card. I created a fresh MS-DOS partition table and erased all existing partition.

```
$ sudo bash
root@lenovo:/home/rick/code/SkiffOS# export PI_SD=/dev/sda
root@lenovo:/home/rick/code/SkiffOS# make cmd/pi/common/format
make[1]: Entering directory '/home/rick/code/SkiffOS/build'
make[2]: Entering directory '/home/rick/code/SkiffOS/configs/pi/common/extensions'
Are you sure? This will completely destroy all data. [y/N] y
Verify that '/dev/sda' is the correct device. Be sure. [y/N] y
+ set -e
+ echo 'Formatting device...'
Formatting device...
+ sudo parted /dev/sda mklabel msdos
Warning: The existing disk label on /dev/sda will be destroyed and all data on this disk will be lost. Do you want to continue?
Yes/No? y                                                                 
Information: You may need to update /etc/fstab.

+ sleep 1                                                                 
+ echo 'Making boot partition...'
Making boot partition...
+ sudo parted -a optimal /dev/sda mkpart primary fat16 0% 300M
Information: You may need to update /etc/fstab.

+ PI_SD_SFX=/dev/sda                                                      
+ '[' -b /dev/sdap1 ']'
+ sudo parted /dev/sda set 1 boot on
Information: You may need to update /etc/fstab.

+ sudo parted /dev/sda set 1 lba on                                       
Information: You may need to update /etc/fstab.

+ echo 'Making rootfs partition...'                                       
Making rootfs partition...
+ sudo parted -a optimal /dev/sda mkpart primary ext4 300M 700MiB
Information: You may need to update /etc/fstab.

+ echo 'Making persist partition...'                                      
Making persist partition...
+ sudo parted -a optimal /dev/sda -- mkpart primary ext4 700MiB -1s
Information: You may need to update /etc/fstab.

+ echo 'Waiting for partprobe...'                                         
Waiting for partprobe...
+ partprobe /dev/sda
+ sleep 2
+ partprobe /dev/sda
+ echo 'Formatting rootfs partition...'
Formatting rootfs partition...
+ mkfs.ext4 -F -L rootfs -O '^64bit' /dev/sda2
mke2fs 1.45.5 (07-Jan-2020)
The file /dev/sda2 does not exist and no size was specified.
make[2]: *** [Makefile:2: format] Error 1
make[2]: Leaving directory '/home/rick/code/SkiffOS/configs/pi/common/extensions'
make[1]: *** [Makefile:10: cmd/pi/common/format] Error 2
make[1]: Leaving directory '/home/rick/code/SkiffOS/build'
make: *** [Makefile:2: cmd/pi/common/format] Error 2
```

paralin, 2021-04-09:
@RickCarlino Are you on Ubuntu by any chance? Note that it creates the partitions with `parted` and then calls `partprobe`, but `/dev/sda2` is still not created. This is actually a Ubuntu bug. Try it with any other distro and `sda3` is created. Not sure why, or how to get Ubuntu to create the partition file after calling `parted`. I've had several people report the same thing.

paralin, 2021-06-29:
Hi @RickCarlino 2021.05 was just released, give it another shot?

You can use the new "make cmd/pi/common/buildimage" command to make a .img if the format script still does not detect the partitions correctly.