skiffos/SkiffOS #150: pinebook pro : nvme utilization?

PlasmaShift, 2021-05-05:
I want to utilize the nvme in my pinebook pro as much as possible but I am unsure the best way to accomplish this.
I would like to boot from the nvme but I can't tell if you can.
You used to be able to boot with a moddified uboot, then mainline got support to boot from the nvme, than mainline dropped support?
If I can't boot from the nvme I would like as much of the OS on the nvme as possible https://wiki.pine64.org/index.php/Pinebook_Pro#Using_as_OS_root_drive

What would the best way to accomplish this with SkiffOS.
This is a cool project reminding me of bedrock linux Genode Guix and NixOs

paralin, 2021-05-05:
hi, welcome. You can probably boot from NVME by simply copying the contents of a SD card formatted with "format" and "install" commands in SkiffOS to the NVME with "dd."

I'd actually recommend instead booting to the emmc:

1. Boot SkiffOS on a SD card
2. Copy the contents of the sd card to the emmc with "dd"
3. Delete the "persist" partition from the emmc
4. Format the NVME with `mkfs.ext4 -F -L persist /dev/pathtonvmehere`
5. Reboot

Skiff will mount anything labeled "persist" as the persist partition for all persistent data + docker images + etc.

PlasmaShift, 2021-05-05:
I got NVME root partition working on Manjaro following https://www.youtube.com/watch?v=8Jcm-0gHC98 which was installed for testing as I want to run Debian or Gentoo over Manjaro.
Now I am  working on compiling SkiffOs and then figuring out SkiffOs

closed 2021-05-05 by PlasmaShift