skiffos/SkiffOS #151: Build fails

tinito, 2021-05-25:
Build fails as [http://sources.buildroot.net/skiff-core-v1.7.2.tar.gz]() doesn't exist (404).
Apparently nothing similar is available on [http://sources.buildroot.net]().

```
--2021-05-25 15:00:08--  http://sources.buildroot.net/skiff-core-v1.7.2.tar.gz
Resolving sources.buildroot.net (sources.buildroot.net)... 2606:4700:20::ac43:4838, 2606:4700:20::681a:25, 2606:4700:20::681a:125, ...
Connecting to sources.buildroot.net (sources.buildroot.net)|2606:4700:20::ac43:4838|:80... connected.
HTTP request sent, awaiting response... 404 Not Found
2021-05-25 15:00:09 ERROR 404: Not Found.

package/pkg-generic.mk:144: recipe for target '/home/tino/tmp/SkiffOS/workspaces/default/build/skiff-core-v1.7.2/.stamp_downloaded' failed
make[3]: *** [/home/tino/tmp/SkiffOS/workspaces/default/build/skiff-core-v1.7.2/.stamp_downloaded] Error 1
Makefile:23: recipe for target '_all' failed
make[2]: *** [_all] Error 2
make[2]: Leaving directory '/home/tino/tmp/SkiffOS/workspaces/default'
Makefile:63: recipe for target 'compile' failed
make[1]: *** [compile] Error 2
make[1]: Leaving directory '/home/tino/tmp/SkiffOS/build'
Makefile:2: recipe for target 'compile' failed
make: *** [compile] Error 2
tino@i7:~/tmp/SkiffOS$ 
```

paralin, 2021-06-08:
@tinito Please give it a try again, this should be fixed as of a while now. Thanks!

closed 2021-06-08 by paralin

paralin, 2021-09-19:
@tinito more fixes for similar issues were merged recently to master