skiffos/SkiffOS #155: Building 2021.02.3 release on RockPro64 for RockPro64 fails

ootoovak, 2021-06-16:
I have RockPro64 running Armbian 4.4y on my RockPro64 SD card right now and I'm trying to build SkiffOS to run on an eMMC module instead (hopefully!).

Trying to compile the 2021.02.3 version of SkiffOS RockPro64 config (on the RockPro64 itself) has not finished successfully though and fails with the following error:
```
/home/cosmonaut/SkiffOS/workspaces/default/host/bin/ccache /usr/bin/gcc -pthread -fPIC -fno-strict-aliasing -O2 -I/home/cosmonaut/SkiffOS/workspaces/default/host/include -DNDEBUG -g -O3 -Wall -Wstrict-prototypes -Ibuild/temp.linux-aarch64-2.7/libffi/include -Ibuild/temp.linux-aarch64-2.7/libffi -I/home/cosmonaut/SkiffOS/workspaces/default/build/host-python-2.7.18/Modules/_ctypes/libffi/src -I. -IInclude -I./Include -I/home/cosmonaut/SkiffOS/workspaces/default/host/include -I/usr/include/aarch64-linux-gnu -I/usr/local/include -I/home/cosmonaut/SkiffOS/workspaces/default/build/host-python-2.7.18/Include -I/home/cosmonaut/SkiffOS/workspaces/default/build/host-python-2.7.18 -c /home/cosmonaut/SkiffOS/workspaces/default/build/host-python-2.7.18/Modules/_ctypes/libffi/src/closures.c -o build/temp.linux-aarch64-2.7/home/cosmonaut/SkiffOS/workspaces/default/build/host-python-2.7.18/Modules/_ctypes/libffi/src/closures.o -Wall -fexceptions
/home/cosmonaut/SkiffOS/workspaces/default/build/host-python-2.7.18/Modules/_ctypes/libffi/src/closures.c: In function 'dlmmap_locked':
/home/cosmonaut/SkiffOS/workspaces/default/build/host-python-2.7.18/Modules/_ctypes/libffi/src/closures.c:460:7: warning: ignoring return value of 'ftruncate', declared with attribute warn_unused_result [-Wunused-result]
  460 |       ftruncate (execfd, offset);
      |       ^~~~~~~~~~~~~~~~~~~~~~~~~~
/home/cosmonaut/SkiffOS/workspaces/default/build/host-python-2.7.18/Modules/_ctypes/libffi/src/closures.c:472:7: warning: ignoring return value of 'ftruncate', declared with attribute warn_unused_result [-Wunused-result]
  472 |       ftruncate (execfd, offset);
      |       ^~~~~~~~~~~~~~~~~~~~~~~~~~
/home/cosmonaut/SkiffOS/workspaces/default/host/bin/ccache /usr/bin/gcc -pthread -fPIC -fno-strict-aliasing -O2 -I/home/cosmonaut/SkiffOS/workspaces/default/host/include -DNDEBUG -g -O3 -Wall -Wstrict-prototypes -Ibuild/temp.linux-aarch64-2.7/libffi/include -Ibuild/temp.linux-aarch64-2.7/libffi -I/home/cosmonaut/SkiffOS/workspaces/default/build/host-python-2.7.18/Modules/_ctypes/libffi/src -I. -IInclude -I./Include -I/home/cosmonaut/SkiffOS/workspaces/default/host/include -I/usr/include/aarch64-linux-gnu -I/usr/local/include -I/home/cosmonaut/SkiffOS/workspaces/default/build/host-python-2.7.18/Include -I/home/cosmonaut/SkiffOS/workspaces/default/build/host-python-2.7.18 -c /home/cosmonaut/SkiffOS/workspaces/default/build/host-python-2.7.18/Modules/_ctypes/libffi/src/aarch64/sysv.S -o build/temp.linux-aarch64-2.7/home/cosmonaut/SkiffOS/workspaces/default/build/host-python-2.7.18/Modules/_ctypes/libffi/src/aarch64/sysv.o -Wall -fexceptions
/home/cosmonaut/SkiffOS/workspaces/default/build/host-python-2.7.18/Modules/_ctypes/libffi/src/aarch64/sysv.S: Assembler messages:
/home/cosmonaut/SkiffOS/workspaces/default/build/host-python-2.7.18/Modules/_ctypes/libffi/src/aarch64/sysv.S:145: Error: constant expression required at operand 2 -- `tbz x23,#AARCH64_FFI_WITH_V_BIT,1f'
/home/cosmonaut/SkiffOS/workspaces/default/build/host-python-2.7.18/Modules/_ctypes/libffi/src/aarch64/sysv.S:174: Error: constant expression required at operand 2 -- `tbz x23,#AARCH64_FFI_WITH_V_BIT,1f'
/home/cosmonaut/SkiffOS/workspaces/default/build/host-python-2.7.18/Modules/_ctypes/libffi/src/aarch64/sysv.S:282: Error: constant expression required at operand 2 -- `tbz x0,#AARCH64_FFI_WITH_V_BIT,1f'
/home/cosmonaut/SkiffOS/workspaces/default/build/host-python-2.7.18/Modules/_ctypes/libffi/src/aarch64/sysv.S:301: Error: constant expression required at operand 2 -- `tbz x0,#AARCH64_FFI_WITH_V_BIT,1f'
/home/cosmonaut/SkiffOS/workspaces/default/build/host-python-2.7.18/Modules/_ctypes/libffi/src/aarch64/sysv.S:260: Error: undefined symbol AARCH64_CALL_CONTEXT_SIZE used as an immediate value

Python build finished, but the necessary bits to build these modules were not found:
_bsddb             _sqlite3           _tkinter
bsddb185           bz2                dbm
dl                 gdbm               imageop
readline           sunaudiodev
To find the necessary bits, look in setup.py in detect_modules() for the module's name.

Failed to build these modules:
_ctypes

make[4]: *** [Makefile:544: sharedmods] Error 1
make[3]: *** [package/pkg-generic.mk:250: /home/cosmonaut/SkiffOS/workspaces/default/build/host-python-2.7.18/.stamp_built] Error 2
make[2]: *** [Makefile:23: _all] Error 2
make[2]: Leaving directory '/home/cosmonaut/SkiffOS/workspaces/default'
make[1]: *** [Makefile:64: compile] Error 2
make[1]: Leaving directory '/home/cosmonaut/SkiffOS/build'
make: *** [Makefile:2: compile] Error 2
```

I was building with the config:
```
export SKIFF_CONFIG=pine64/rockpro64,core/gentoo,apps/yacht
```

I'll try to debug this myself but any help would be appreciated. 😃 

Also, I really hope this will boot from the eMMC module. So far I've had no success finding a Linux with a 5.x kernel that can do it and I think I've tried all the ones listed on the Pine64 website! 😞

ootoovak, 2021-06-16:
It seems like it might be a Python build requirement that my system currently does not have. What are the pre-build requirements for SkiffOS?

I'll report back if I found out what this one is.

ootoovak, 2021-06-16:
According to the command:
```
sudo apt-get build-dep python2.7
```

The dependancies are:
```
autopoint blt-dev debhelper dh-autoreconf dh-strip-nondeterminism diffstat docutils-common dwz fontconfig-config
  fonts-dejavu-core gettext gettext-base help2man intltool-debian libarchive-zip-perl libbluetooth-dev libbz2-dev
  libcroco3 libdb-dev libdb5.3-dev libdebhelper-perl libdrm-amdgpu1 libdrm-common libdrm-nouveau2 libdrm-radeon1
  libdrm2 libexpat1-dev libfile-stripnondeterminism-perl libfontconfig1 libfontconfig1-dev libfontenc1
  libfreetype-dev libfreetype6 libfreetype6-dev libgdbm-dev libgl1 libgl1-mesa-dri libglapi-mesa libglvnd0
  libglx-mesa0 libglx0 libice-dev libice6 libjs-jquery libjs-sphinxdoc libjs-underscore libllvm11 libncurses-dev
  libncursesw5-dev libpixman-1-0 libpng-dev libpthread-stubs0-dev libreadline-dev libsm-dev libsm6 libsqlite3-dev
  libsub-override-perl libtk8.6 libtool libunwind8 libx11-6 libx11-data libx11-dev libx11-xcb1 libxau-dev libxau6
  libxaw7 libxcb-dri2-0 libxcb-dri3-0 libxcb-glx0 libxcb-present0 libxcb-sync1 libxcb-xfixes0 libxcb1 libxcb1-dev
  libxdamage1 libxdmcp-dev libxdmcp6 libxext-dev libxext6 libxfixes3 libxfont2 libxft-dev libxft2 libxkbfile1
  libxmu6 libxmuu1 libxpm4 libxrender-dev libxrender1 libxshmfence1 libxss-dev libxss1 libxt-dev libxt6
  libxxf86vm1 pkg-config po-debconf python-babel-localedata python3-alabaster python3-babel python3-docutils
  python3-imagesize python3-jinja2 python3-markupsafe python3-packaging python3-pygments python3-pyparsing
  python3-roman python3-sphinx python3-tz quilt sgml-base sharutils sphinx-common tcl tcl-dev tcl8.6-dev time tk
  tk-dev tk8.6 tk8.6-blt2.5 tk8.6-dev uuid-dev x11-common x11-xkb-utils x11proto-core-dev x11proto-dev
  x11proto-scrnsaver-dev x11proto-xext-dev xauth xml-core xorg-sgml-doctools xserver-common xtrans-dev xvfb
  zlib1g-dev
  ```
  
  I'm sure I don't have to install them all like that (I'm sure some are transitive dependancies) but I might have to give that a go!

ootoovak, 2021-06-16:
Ok, after running the above it still left these in the error:
```
Python build finished, but the necessary bits to build these modules were not found:
bsddb185           dl                 imageop
sunaudiodev
To find the necessary bits, look in setup.py in detect_modules() for the module's name.

Failed to build these modules:
_ctypes
```

So, there are fewer but I'll need to track down the dependancies for each.

_(Recording here just in case it helps anyone else.)_

ootoovak, 2021-06-16:
Ok, I've giving up for now, seems like this might be an issue with trying to cross compile Python on an ARM machine. 😞

paralin, 2021-06-16:
I've run into this before with the arm workers for the GitHub actions, at a certain revision it just started working - try "master" latest with "git submodule update"

Your host system for building appears to be Ubuntu or Debian, If so, it might be like you said, missing dependencies or something outdated.

It'd be good to find out why, perhaps there's some fix that can be done in the Buildroot side of things for backwards compat.

ootoovak, 2021-06-16:
_(Yeah, this version of Armbian is based off of Ubuntu.)_

Sadly, the main branch also fails with the same error. It could be a Python thing too, maybe? I wonder if that module can only be cross compiled from x86. I'll see if I can find out.

Otherwise, I should start learning some Buildroot. 😅

paralin, 2021-06-16:
Well, the issue is definitely with building host python on arm systems - and the issue lies with Buildroot, unfortunately, not skiff. I can always import a fix into the skiff Buildroot tree if one is found.

I had previously closed my bugzilla entry for this, will reopen it.

paralin, 2021-06-17:
Confirming same issue as #142

paralin, 2021-06-17:
@ootoovak In the meantime, you can compile the OS on a Intel host (or AMD) and Buildroot will take care of cross compiling everything for you.

closed 2021-06-17 by paralin

paralin, 2021-06-18:
@ootoovak https://bugs.busybox.net/show_bug.cgi?id=13661&GoAheadAndLogIn=1

There is a patch posted there which fixes it apparently. I have applied it to the skiff "next" (Buildroot 2021.05) or "staging" (master) branches, can you give it a try on either of those?

paralin, 2021-09-19:
We believe this is now fixed as of Buildroot commit b31f7bc958.