skiffos/SkiffOS #158: Build error witth kernel > 5.12 -  fatal error: linux/cyclades.h: No such file or directory

s1ng0c, 2021-06-23:
Hi Team,

I am trying to build image for pine64/phone which currently uses kernel 5.13. I got error due to removal of **linux/cyclades.h** on Linux kernel itself. Could you please check?

The issue is under discussion in [here](https://www.mail-archive.com/gcc@gcc.gnu.org/msg95386.html)

paralin, 2021-06-23:
@s1ng0c the error is when building which package, libsanitize?

s1ng0c, 2021-06-23:
One other issue you might wanna check (cuz I don't have log so I could't create new issue) where *http://sources.buildroot.net/embiggen-disk* is not existed anymore(seem) but still **buildroot** is somehow reference to it. Download error hence the build was broken.

paralin, 2021-06-24:
@s1ng0c Most likely you had issues getting to the Github download server, embiggen-disk is using the following URL to download (the sources.buildroot is the fallback): https://github.com/bradfitz/embiggen-disk/archive/c554fc1c93a4004ce0b6a3f69b0dabe0481a2308/embiggen-disk.tar.gz

paralin, 2021-06-24:
Fixed by disabling libsanitizer in gcc on the skiff buildroot tree.

closed 2021-06-24 by paralin

s1ng0c, 2021-06-24:
> @s1ng0c Most likely you had issues getting to the Github download server, embiggen-disk is using the following URL to download (the sources.buildroot is the fallback): https://github.com/bradfitz/embiggen-disk/archive/c554fc1c93a4004ce0b6a3f69b0dabe0481a2308/embiggen-disk.tar.gz

OK, let me try one more