skiffos/SkiffOS #163: Can we by any chance enable jobserver to run multiple jobs?

marodgz, 2021-09-15:
Trying this command: `$ time make -j 9 configure 2>&1 > configure_.log`

and getting this warning:
```make[1]: warning: jobserver unavailable: using -j1.  Add '+' to parent make rule.```

Regards

paralin, 2021-09-15:
@marodgz It uses multiple jobs - you can set the quantity with `BR2_JLEVEL`:

```
make compile BR2_JLEVEL=9
```

Normally it defaults to the number of CPU cores you have + 1.

If that's not working it has to be something related to Make on your system. See: https://buildroot.org/downloads/manual/manual.html section 8.12

marodgz, 2021-09-15:
@paralin It seems to be working after I passed the BR2_LEVEL variable. I wonder how can I speed up the compile process cause last time I tried directly in my Pinephone it took around 8 hrs. Now I'm trying that in my laptop, and planning to flash the image using Jumpdrive.

Is that the expected workflow? Any other suggestion?

paralin, 2021-09-15:
@marodgz It's optimized for cross compile, use a fast machine to build the OS and then flash it using Jumpdrive (as you said).

I'm surprised you actually managed to build it all directly on the Pinephone. That will work, it'll just be slow. And you can update the system that way too - just `./scripts/push_image.sh root@mypinephone` once its running skiff already

closed 2021-09-17 by paralin