skiffos/SkiffOS #164: Can't fetch embiggen-disk from sources.buildroot.net

marodgz, 2021-09-15:
make compile throws this error: ```ERROR: embiggen-disk-c554fc1c93a4004ce0b6a3f69b0dabe0481a2308.tar.gz has wrong sha256 hash:
ERROR: expected: cf67e8dea3db7d0743a4a530695e1341ac9c1c9cf773bd157bdb94f069615485
ERROR: got     : 3558e1f3c2b8ceaef58c7313c6f2b5504c8130daca1628d5af1bbc1144048e06
ERROR: Incomplete download, or man-in-the-middle (MITM) attack
--2021-09-15 17:50:15--  http://sources.buildroot.net/embiggen-disk/embiggen-disk-c554fc1c93a4004ce0b6a3f69b0dabe0481a2308.tar.gz
Resolving sources.buildroot.net (sources.buildroot.net)... 104.26.1.37, 172.67.72.56, 104.26.0.37, ...
Connecting to sources.buildroot.net (sources.buildroot.net)|104.26.1.37|:80... connected.
HTTP request sent, awaiting response... 404 Not Found
2021-09-15 17:50:15 ERROR 404: Not Found.
```

The hash seems to be valid: https://github.com/bradfitz/embiggen-disk/commit/c554fc1c93a4004ce0b6a3f69b0dabe0481a2308

However there's no folder called embiggen-disk here http://sources.buildroot.net/

paralin, 2021-09-19:
@marodgz This is due to the "go mod vendor" step producing non-deterministic output.

Buildroot falls back to downloading from their sources website when the hash is wrong (as it is here).

paralin, 2021-09-19:
I submitted this issue to the Buildroot mailing list a long time ago: https://yhbt.net/lore/all/20210801091416.GQ3189549@scaer/

Summary:

The access modes are different: In the 'good' archive, it is 0644 (rw-r--r--), while in the 'bad' archive, it is 0664 (rw-rw-r--). For some reason, Buildroot is not enforcing this mode properly when creating the .tar.gz (on some systems).

I'll ping the list again and see if they have any solution.

paralin, 2021-09-19:
@marodgz I found a fix, it now uses a different archive format, please try latest "master" and be sure to "go submodule update" to get the new submodule. It should work with "make compile" right away.

paralin, 2022-02-12:
Seems hash might be wrong again

commit dfbee731 (2021-09-19): buildroot: fix go mod vendor repack determinism

Fixes #164
commit fd09ea30 (2022-02-13): configs-base: add skiffos mirror as default backup site

Change from the sources.buildroot.net mirror to:

https://github.com/skiffos/mirror

And add embiggen-disk.

Fixes #164
commit 5b5d0f19 (2022-03-11): configs-base: add skiffos mirror as default backup site

Change from the sources.buildroot.net mirror to:

https://github.com/skiffos/mirror

And add embiggen-disk.

Fixes #164
closed 2022-02-13 by paralin