skiffos/SkiffOS #165: Missing information on why certain systems are discontinued and what it means

maxromanovsky, 2021-09-19:
I haven't found any detailed info on discontinued status for systems:

1. What it actually means (is it still possible to use? Are there any old builds available? Can newer builds be compiled?)
2. When the support will be dropped entirely
3. What is the reason for being discontinued?
4. Any plans to discontinue other boards?

For instance, I have an Odroid C2 (for which Armbian is available BTW), and a bunch of RPI 3, and I am not sure if it's worth trying SkiffOS on then. Previously I was running HypriotOS, which got abandoned.

paralin, 2021-09-19:
Discontinued by the manufacturer, not by Skiff. Probably could be made clearer - could there be a better term to put there in the notes field, preferably with the same or less characters?

paralin, 2021-09-19:
> ODROID-C2 will be discontinued due to the CPU (Amlogic's S905) supplying discontinuation

paralin, 2021-09-19:
How about "Obsolete" - definition: no longer produced or used; out of date.

maxromanovsky, 2021-09-19:
I think there's no ideal way to describe it concisely in the table. I am totally fine with both "Discontinued" and "Obsolete", as long as there's more detailed explanation/footnote above/below the table. Maybe even with link to this issue.

For instance:
> Board was discontinued by the manufacturer. It is still supported as the SkiffOS target. See #165 for more details.

paralin, 2021-09-19:
Ok added with 94c8048f00c7318a7d6ca52ff50ad8178870bb5b and 3403fbbe2bd71984df534232b7f269239897347b, thanks.

closed 2021-09-19 by paralin

paralin, 2021-09-19:
>and a bunch of RPI 3, and I am not sure if it's worth trying SkiffOS on then. Previously I was running HypriotOS, which got abandoned.

By the way - pi3 is fully supported & tested, worth a try.

maxromanovsky, 2021-09-19:
@paralin thank you. That's the plan ;)