skiffos/SkiffOS #171: Rootfs as a file?

bam80, 2021-12-12:
Hello,
does rootfs install to a file and mount as loopback device from there or it's extracted to a whole dedicated partition finally?
It doesn't quite clear from `push_image.bash` script.

bam80, 2021-12-12:
Could you describe initial system partition layout? Particularly, for PinePhone.

paralin, 2021-12-14:
@bam80 hi, the system boots to either an initramfs (in-memory copy of the SkiffOS build) or a .squashfs mount, depending on the target. (Pi uses initramfs, Jetson tx2 uses squashfs, for example).

The initial system partition layout for pinephone is:

 - boot: contains the .dtb, zimage, initramfs, etc.
 - rootfs: contains the modules.squashfs with kernel modules
 - persist: contains the persistent data (initially empty)

bam80, 2021-12-15:
> or a .squashfs mount

Thanks. So the rootfs is loopback-mounted from that .squashfs file placed on some other fs, and not expanding to the whole rootfs partition?  

That would be great, do you think it is possible to install it in parallel with some other OS (e.g Manjaro ARM) without initial formatting all the device?

paralin, 2021-12-15:
@bam80 many single-board computers (pi) boot from the first partition and I'm not sure if you can specify an alternate to boot from.

However some of them - i.e. the ones that use u-boot like pinephone - search for boot.scr or extlinux.conf, so possibly that might be able to dual-boot.

bam80, 2021-12-15:
E.g. Manjaro uses following partitions layout for PP:
- boot (VFAT) for u-boot+scripts, zImage, DTBs and such
- root (EXT-4) for rootfs, /home etc.

So if one wanted to try install SkiffOS in parallel, basically he should follow the installation scripts and carefully unpack the images to appropriate places, and edit u-boot scripts afterwards?
Should he name the partitions in a proper way, and create `persist` partition additionally?

Can u-boot be shared between systems?

paralin, 2021-12-15:
You could make a single partition named "persist" and place the contents of the "boot" partition there. (See install_sd.sh).

The problem is some of the u-boot scripts use "ext4load" and "fatload" and some others assume that Skiff uses the 3-partition layout.

So it depends per-platform. Any platform that uses extlinux.conf should work.

There's a config used by jetson/tx2 which boots to a squashfs. This one puts both the persist and the /boot files into a single persist partition. So, using this type of setup for other platforms might also make it possible to multi-boot.

paralin, 2021-12-15:
... That said for now I would just recommend having 1 SD card per OS you want, rather than trying to mix multiple on a single. You can just import your other OS (like manjaro) as a Skiff Core container.

bam80, 2021-12-15:
> You can just import your other OS (like manjaro) as a Skiff Core container.

I'm quite intrigued how would it work. Does it have full access to the hardware from the container - GUI, Modem, etc? What differences from vanilla Manjaro can I expect?

paralin, 2021-12-15:
@bam80 You can run "tar -c . | docker import - my/image" to import a existing system rootfs to a image.

Then you can set it as the Skiff Core image and it will be started as a container.

The isolation features of the container are disabled so all hardware is available.

You will want to disable ModemManager in the container (or in Skiff) with systemctl mask ModemManager because otherwise there will be two of them running. Also, you probably want to mask NetworkManager in the container since Skiff runs an instance of that too.

bam80, 2021-12-15:
> You will want to disable ModemManager in the container (or in Skiff) with systemctl mask ModemManager because otherwise there will be two of them running. Also, you probably want to mask NetworkManager in the container since Skiff runs an instance of that too.

Hmm, quite important note. Is it documented somewhere?

paralin, 2021-12-16:
Probably not documented anywhere yet. The normal skiff core containers mask those things in their dockerfiles.

closed 2022-01-26 by paralin