skiffos/SkiffOS #172: conflicting services between SkiffOS and containers

bam80, 2021-12-16:
Probably should be documented at least, if we can't come up with some generic solution.

> You will want to disable ModemManager in the container (or in Skiff) with systemctl mask ModemManager because otherwise there will be two of them running. Also, you probably want to mask NetworkManager in the container since Skiff runs an instance of that too.

https://github.com/skiffos/SkiffOS/issues/171#issuecomment-995222511

paralin, 2021-12-17:
Specifically for "Skiff Core" the containers are configured (in skiff-core.yaml) to have no isolation (--privileged --net=host etc).

Then, full OS images are run as containers - usually running systemd as pid 1 in the container.

This means that all processes inside the container will have equal authority over the hardware as the "host OS" processes. So, any duplicate service running both in the container and in the host OS could cause issues. 

In this case, ModemManager and NetworkManager definitely can't be run multiple instances simultaneously without causing issues.

bam80, 2021-12-17:
Could we move the text from here to a more proper place, maybe?

paralin, 2021-12-19:
@bam80 Yeah I think soon it will be time to develop a proper documentation website for Skiff (like readthedocs, etc).

closed 2022-03-09 by paralin