skiffos/SkiffOS #178: ssh root@my-ip-address logon on hc1

devbourne, 2022-02-20:
I build twice for odroid-hc1

First,
$ export SKIFF_CONFIG=odroid/xu
$ make configure                   
$ make compile                    
$ sudo bash             
$ export ODROID_SD=/dev/sdb
$ make cmd/odroid/common/format  
$ make cmd/odroid/common/install 
$ ssh root@192.168.0.48 (in my case)

but, it requires password.

So, I did build again
$ export SKIFF_CONFIG=odroid/xu,skiff/core
$ make configure                  
## this time, I ran menuconfig  to add samba4 service
$ make br/menuconfig            
$ make compile                     
$ sudo bash             
$ export ODROID_SD=/dev/sdb
$ make cmd/odroid/common/format  
$ make cmd/odroid/common/install 
$ssh root@192.168.0.48 (in my case)

It requires password, too.

paralin, 2022-02-23:
Hi there @devbourne 

You are doing everything correctly, however, you need to add your SSH key to the machine.

You can generate one with ssh-keygen (press enter repeatedly to use defaults).

Your public key is ~/.ssh/id_rsa.pub

You can either mount the persist partition - sdb3 (3rd partition) - and place your key under persist/skiff/keys/my-key.pub

Or you can place my-key.pub before running "make compile" at ./overrides/root_overlay/etc/skiff/authorized_keys/my-key.pub and it will also be accepted for root.

paralin, 2022-02-27:
Feel free to ping here or join Discord if you need more help.

closed 2022-02-27 by paralin