skiffos/SkiffOS #183: jetson-nano: running skiff-jetson-4.9.x on Jetpack 4.6.1 (Ubuntu 20.04.4)

braveheartleo, 2022-03-21:
Hi @paralin 

I appreciate the work you did bringing up the L4T Tegra Kernel release of nVidia from 4.9.253 to 4.9.306. I managed to compile the and run this kernel on the Nano under the Jetpack 4.6.1 that was upgraded from Ubuntu 18.04.6 bionic to 20.04.4 focal.

The first compile I tried with gcc-9 with tegra_defconfig and it was successful, albeit there were some warnings about some changes on the gcc-9 release. I tried gcc-7 that is the recommended version from nVidia and it was successful as well.

On both instances I noticed something, although non-fatal, from dmesg that is different from the nVidia release,

```
[    0.000000] Booting Linux on physical CPU 0x0
[    0.000000] Linux version 4.9.306-tegra-custom (user@jetson-nano) (gcc version 9.4.0 (Ubuntu 9.4.0-1ubuntu1~20.04) ) #1 SMP PREEMPT Sat Mar 19 14:58:42 PST 2022
[    0.000000] Boot CPU: AArch64 Processor [411fd071]
[    0.000000] OF: fdt:memory scan node memory@80000000, reg size 32,
[    0.000000] OF: fdt: - 80000000 ,  7ee00000
[    0.000000] OF: fdt: - 100000000 ,  7f200000
[    0.000000] earlycon: uart8250 at MMIO32 0x0000000070006000 (options '')
[    0.000000] bootconsole [uart8250] enabled
[    0.000000] OF: fdt:Reserved memory: failed to reserve memory for node 'fb0_carveout': base 0x0000000000000000, size 0 MiB
[    0.000000] OF: fdt:Reserved memory: failed to reserve memory for node 'fb0_carveout': base 0x0000000000000000, size 0 MiB
[    0.000000] OF: fdt:Reserved memory: failed to reserve memory for node 'fb1_carveout': base 0x0000000000000000, size 0 MiB
[    0.000000] OF: fdt:Reserved memory: failed to reserve memory for node 'fb1_carveout': base 0x0000000000000000, size 0 MiB
```

I was wondering if you have any idea regarding those `failed to reserve memory for node` messages.

Many thanks once again for bringing this aging kernel release up -to-date.

paralin, 2022-03-21:
@braveheartleo There's always tons of errors in kernel logs, even on perfectly working systems.

I wouldn't worry about it too much unless you notice something actually broken.

That log message appears in a lot of normal jetson nano dmesg logs online that I've seen.

paralin, 2022-03-21:
That said, I'd definitely recommend running SkiffOS and then running Ubuntu 20.04.4 Focal inside a container.

If you run with jetson/nano,skiff/core you will get this automatically w/ Jetpack in the container.

braveheartleo, 2022-03-21:
> That said, I'd definitely recommend running SkiffOS and then running Ubuntu 20.04.4 Focal inside a container.
> 
> If you run with jetson/nano,skiff/core you will get this automatically w/ Jetpack in the container.

Yes, there's nothing broken so far I that I have observed.

I'll try SkiffOS. I first came upon your comment on being succesful in merging 306 into the Jetson kernel here https://github.com/OE4T/linux-tegra-4.9/issues/21 that lead me here.  Nice project btw. 😊

closed 2022-03-21 by braveheartleo

paralin, 2022-03-21:
@braveheartleo Thanks! & I just merged it to 4.9.307 :)