skiffos/SkiffOS #185: podman: support and consider migrating skiff-core

paralin, 2022-03-23:
podman is a modern lighter weight alternative to Docker that has the same CLI and supports features like rootless containers.

It supports using the "crun" c-based OCI runtime which has a smaller memory overhead than runc (used today).

Docker would still be supported with the apps/docker flag, but skiff/core would now use apps/podman instead.

Skiff/core would be rewritten to use the podman API instead of the Docker API.

paralin, 2022-03-23:
Could it make sense to use Podman instead of Docker?

Docker would still be supported with the apps/docker flag, but skiff/core would now use apps/podman instead.

podman is a modern lighter weight alternative to Docker that has the same CLI and supports features like rootless containers.

Skiff/core would be rewritten to use the podman API instead of the Docker API.

Curently, I think the downsides of switching away from containerd outweigh the advantages.

tianyuanhao, 2022-11-07:
>> Could you please review and test [this patch](http://patchwork.ozlabs.org/project/buildroot/patch/20221006140754.46144-1-tianyuanhao3@163.com/)?
>
> With that patch I can just deselect `runc` and enable `crun` instead?
> But I think _Docker_ also selects `runc`. Might need a few more conditional statements.

@paralin You can deselect `runc` and select `crun`. _Docker_ doesn't select `runc` but `containerd`.

paralin, 2022-11-07:
Hey @tianyuanhao I'll try it out tonight, thanks for the reminder!

tianyuanhao, 2022-11-21:
Let's get it merged before `2023.02`.

paralin, 2022-11-21:
@tianyuanhao is it possible to add a config file for containerd to change it to use CRUN, as opposed to the symlink? I'll add my reviewed-by to the patch in the mailing list but they might not like the symlink approach

tianyuanhao, 2022-11-21:
@paralin The _config approach_ is a bit complicated for `Buildroot`. The relationship between `crun` and `runc` is like `busybox` and `coreutils`, so I use _symlink approach_.

paralin, 2022-11-22:
@tianyuanhao I submitted my reviewed-by on the mailing list and have merged your commit into the skiffos tree.

I'll try now to switch runc for crun on all skiffos default configs. Thanks!

paralin, 2023-07-19:
This is now integrated.

closed 2023-07-19 by paralin