skiffos/SkiffOS #186: podman: optionally use instead of docker for skiff-core

paralin, 2022-03-23:
Add support for podman under the apps/podman layer.

podman is a modern lighter weight alternative to Docker that has the same CLI and supports features like rootless containers.

Skiff/core could be rewritten to use the podman API instead of the Docker API, and use a k8s Pod syntax to define containers instead of the skiff-core yaml config used today.

Greylinux, 2022-04-01:
I was just thinking about this , this morning and I thought i'll create an issue for it.  Beat me to the punch ! 
Nice work @paralin

tianyuanhao, 2022-04-07:
@paralin It feels complicated, you can refer to https://github.com/kubernetes/minikube/tree/master/deploy/iso/minikube-iso/package.

paralin, 2022-07-24:
https://github.com/skiffos/buildroot/pull/20

Added a working podman package for both host & target.

There are a few TODOs remaining before merging.

VarLad, 2022-07-31:
@paralin After podman, can I make a feature request for https://github.com/containers/toolbox and https://github.com/89luca89/distrobox ?
Those would be convenient to have on SkiffOS

paralin, 2022-08-01:
@VarLad can do. In the meantime, though, you can probably accomplish the same thing with Skiff Core.

aschamberger, 2022-08-20:
> In future:
> 
> Skiff/core could be rewritten to use the podman API instead of the Docker API, and use a k8s Pod syntax to define containers instead of the skiff-core yaml config used today.

Another alternative would be to implement the docker compose yaml format. This way you could just use docker compose on another OS and later migrate to SkiffOS. 

This would help in migrating my project to SkiffOS: I use docker compose on PiOS for my multiroom audio server to set up 8 container instances with all the container specific config within an env file. Via the profile feature I can only start the desired instances via `docker-compose -f compose.yml --profile on up -d` based on environment variables. I plan to build sometime a small supervisor container that allows changing the env file.

paralin, 2022-08-20:
@aschamberger You can already use docker compose with `apps/compose` - Skiff Core is more meant to define long-running containers with ssh drop-in.

I don't know if the compose format would translate 1-1

paralin, 2022-08-21:
All of the Buildroot packages for podman, catatonit, conmon, cni-plugins, crun have been submitted upstream & are in the SkiffOS tree. #235 tracks merging the config packages `apps/podman` and `apps/crun`

paralin, 2022-08-21:
Podman support is merged to master with `apps/podman` 🥳

closed 2022-08-21 by paralin

tianyuanhao, 2022-08-22:
Hey @paralin, that's wonderful. I'm following [the relevant patches in Buildroot](http://patchwork.ozlabs.org/project/buildroot/list/?submitter=66856). I recommend that you refer to [these files in minikube](https://github.com/kubernetes/minikube/tree/master/deploy/iso/minikube-iso/package) and send v2 patches.

[Download the `crun` source package instead of using `git`](https://github.com/kubernetes/minikube/blob/master/deploy/iso/minikube-iso/package/crun/crun.mk#L9-L10) so no hacking is required.

VarLad, 2022-08-22:
> @VarLad can do. In the meantime, though, you can probably accomplish the same thing with Skiff Core.

@paralin Any chance we can get a Fedora/Fedora-based image in skiff/core in the future? :sweat_smile:

paralin, 2022-08-23:
@tianyuanhao I updated the package, thanks. Will resubmit.

By the way, some of those Buildroot packages in minikube could use an update. Replace `cni` with my `cni-plugins` for example, because `cni` doesn't even use the Go build infrastructure. The licenses & a few other details could be updated in `crun` too.

Hopefully they just use the upstream packages once they are merged.

@VarLad please open an issue on that

paralin, 2022-08-23:
@tianyuanhao For the update... We still need to run AUTORECONF...

```
WARNING: 'automake-1.16' is probably too old.
         You should only need it if you modified 'Makefile.am' or
         'configure.ac' or m4 files included by 'configure.ac'.
         The 'automake' program is part of the GNU Automake package:
```

paralin, 2022-08-23:
@VarLad Fedora, distrobox, and toolbx are all added. :)

VarLad, 2022-08-23:
That was really fast. Thanks a lot @paralin

tianyuanhao, 2024-07-25:
In fact, podman has not been merged into the Buildroot master branch yet.

paralin, 2024-07-25:
@tianyuanhao that's true. It needs some love, I'm still using containerd and runc and don't have time to switch to podman right now. If you want to fix it up, PRs are welcome! And feel free to resubmit my patches to buildroot with your revisions.