skiffos/SkiffOS #192: Kernel panic on Raspberry Pi 2 Model B v1.1

yokoyama-flogics, 2022-04-19:
Hello,

Thank you for the great software.

After reading [this](https://github.com/skiffos/SkiffOS/tree/master/configs/pi), I thought configuration `pi/3` should work on Raspberry Pi 2 Model B v1.1 (revision code from cpuinfo is `a01041`.)

Used SkiffOS version is 2022.02.1.  My procedure was:

```bash
$ cd SkiffOS
$ make
$ export SKIFF_WORKSPACE=default
$ export SKIFF_CONFIG=pi/3,skiff/core
$ make configure
$ make compile
$ sudo bash
# export SKIFF_WORKSPACE=default
# export PI_IMAGE=./pi-image.img
# make cmd/pi/common/buildimage
```

Here is the kernel [boot log](https://github.com/skiffos/SkiffOS/files/8510836/panic.log).

On the other hand, I confirmed that the same build image ran on Raspberry Pi 3 Model B Plus Rev 1.3 (hardware revision is `a020d3` from cpuinfo).

Thank you for any advises.

Regards,
Atsushi

paralin, 2022-04-19:
Hi, thanks for the report, I guess Pi 2 needs a separate config. 

I added `pi/2` on the latest master commit 7091461c58f604edcb9b46d15210e0d282fd16cf

You'll need to rebuild with pi/2 instead which uses a build for the older CPU.

Please feel free to re-open / reply here if this doesn't fix it.

closed 2022-04-19 by paralin

yokoyama-flogics, 2022-04-19:
Hello @paralin,

Thank you for your prompt reply!

Yes, I will try that.

By the way, in generally speaking, when we change `SKIFF_CONFIG`, do we need `make clean` or similar?  Why I ask is, I noticed `make compile` requires quite long time to complete.  (More than 4 hours on my old PC.)

In this time case, we need to check out the different commit from Git, I guess we need a new `SKIFF_CONFIG` and `make compile` in anyway...

Thanks and Regards,
Atsushi