skiffos/SkiffOS #193: I have several presently unsupported devices

jim255, 2022-04-20:
Wandboard
Banana Pi M2 (original Allwinner A31s)
Beagle boards running TI AM3358 & AM5729
And a yet to be released Beagle based on TI J721e (ARM64)
...
I guess I'll try to figure out the configuration...
-Jim

jim255, 2022-04-20:
I'll gladly debug, when ready to test!
-Jim

paralin, 2022-04-21:
Added `bananapi/m2` configuration layer.

paralin, 2022-04-21:
Added `freescale/wandboard` configuration layer.

paralin, 2022-04-22:
@jim255 looking forward to hearing how those two configs work for you. Let me know if you'd like a precompiled image - I can make & upload one for you.

jim255, 2022-04-23:
I'm having difficulties building.   this is the error I'm seeing.
make[1]: Entering directory '/home/janderso/Desktop/SkiffOS-master/build'
fatal: not in a git directory
Failed to register Git Hooks.
fatal: not a git repository (or any of the parent directories): .git
make[1]: *** [Makefile:20: ensure-submodule] Error 128
make[1]: Leaving directory '/home/janderso/Desktop/SkiffOS-master/build'
make: *** [Makefile:2: configure] Error 2
What am I doing wrong?

paralin, 2022-04-23:
@jim255 please use git to clone the repo instead of downloading it. 

"git clone https://github.com/skiffos/skiffos"

paralin, 2022-04-26:
@jim255 That log doesn't show anything, could you instead send the skiff build output?

jim255, 2022-04-26:
Wandboard also won't build.
[wandboard.build.log](https://github.com/skiffos/SkiffOS/files/8564229/wandboard.build.log)

paralin, 2022-04-26:
@jim255 It seems you don't have gcc installed.

You will need to install it:

```
apt-get install build-essential
```

jim255, 2022-04-26:
I also added suggestions into the Beagleboard community to have a look here.. Is there a command or process to build an (.img) image file without using an sd-card?

paralin, 2022-04-26:
Yes, for example make cmd/pi/common/buildimage

That will allocate a file on disk, mount it with a loop device, then run the usual format + install commands against the loop device, then unmount the loop device.

This just makes the kernel emulate a sd card with a file.

paralin, 2022-04-26:
@jim255 The beagleboard config for beaglebone ai is available at: https://github.com/skiffos/SkiffOS/pull/123

I need to build + test it still.

jim255, 2022-04-26:
latest build failure freescale/wandboard config.

excerpt
t
>>> uboot 2022.04 Installing to images directory
cp -dpf /home/janderso/skiffos/workspaces/wandboard/build/uboot-2022.04/u-boot.bin /home/janderso/skiffos/workspaces/wandboard/images/
cp -dpf /home/janderso/skiffos/workspaces/wandboard/build/uboot-2022.04/u-boot-sunxi-with-spl.bin /home/janderso/skiffos/workspaces/wandboard/images/
cp: cannot stat '/home/janderso/skiffos/workspaces/wandboard/build/uboot-2022.04/u-boot-sunxi-with-spl.bin': No such file or directory
make[3]: *** [package/pkg-generic.mk:373: /home/janderso/skiffos/workspaces/wandboard/build/uboot-2022.04/.stamp_images_installed] Error 1
make[2]: *** [Makefile:23: _all] Error 2
make[2]: Leaving directory '/home/janderso/skiffos/workspaces/wandboard'
make[1]: *** [Makefile:64: compile] Error 2
make[1]: Leaving directory '/home/janderso/skiffos/build'
make: *** [Makefile:2: compile] Error 2

paralin, 2022-04-26:
@jim255 please pull "master" and "make compile" to resume the build.

jim255, 2022-04-26:
I did that, it's still referencing the non existent  u-boot-sunxi-with-spl.bin 
????

paralin, 2022-04-26:
Ah, please also `rm -rf ./workspaces/wandboard/build/uboot-2022.04` @jim255

jim255, 2022-04-26:
still no joy.....

paralin, 2022-04-26:
@jim255 Just to confirm you've pulled latest master, deleted build/uboot-2022.04, and "make configure compile"?

I'll run a build on my end to reproduce it.

jim255, 2022-04-26:
twice!

paralin, 2022-04-26:
@jim255 OK, I found a issue with how uboot was configured, please pull & delete u-boot & make configure compile once more.

paralin, 2022-04-26:
I pushed one more commit 14f892dd8f2a68b80dfe854f9dc8807fd5a1143a to fix the install script.

The build completed successfully for me this time.

paralin, 2022-04-27:
@jim255 the beagleboard builds are ready at #123 however from my testing on a beagleboard ai, it ignores the sd card and boots from internal memory instead. I don't have a serial debug cable (ordered one) to debug that currently. Will post updates in the PR.

jim255, 2022-04-30:
There are 2 buttons next to the network connector. Press both to reset, release order determines the persistent boot device setting.

paralin, 2022-05-10:
@jim255 Works now correctly, `beaglebone/ai` merged to master.

paralin, 2022-05-21:
hey @jim255 - been using the beaglebone ai w/ skiff for a week or so now, works fine! looking forward to hearing if you've had a chance to test.

closed 2022-05-22 by paralin