skiffos/SkiffOS #195: booting problem rpi3

Lwing, 2022-04-25:
I've been trying to boot rpi3 for a few days now always get the same error:  systemd-gpt-auto-generator failed with 1.
Then systemd-modules-load fails because of this 
Then I get a few audit traces
Then nothing
I've tried a bunch different configurations. Pi/3,skiff/core,apps/yacht is the one I want.
I haven't tried any other base is though (Gentoo, Ubuntu, etc).
Also, I've tried both booting from SD card and usb, no difference.
I really dislike yocto and this project seems perfect for my current requirements - simple and really powerful.
Am I doing something wrong? What config should work?

paralin, 2022-04-25:
Hmm, I have several pi 3 working fine here. 

Are you sure the boot failed? because it's not going to show a GUI on default, just boot logs.

Happy to help debug it, we will figure it out in no time.

paralin, 2022-04-25:
So the Pi 3 expects the following partition layout:

 - boot: contains the pi3 boot files, rootfs.cpio.gz (fat32)
 -  rootfs: contains modules.squashfs (ext4)
 - persist: storage for persistent data (usually ext4), empty on first boot

Lwing, 2022-04-26:
Yes, ofc. The flash layout seems fine. I don't think I'd get systemd errors if there was a problem with partitioning or boot files. Right?
I added init=/bin/bash to the command line, but the try doesn't come up. 
Which config do you normally use?
Thank you for your extremely fast and helpful answers

paralin, 2022-04-26:
Please remove the init arg. On raspberry pi, the init should be /dev/ram0 or /dev/initrd

Yes indeed, systemd logs would show that it managed to start systemd. I ask because there's a step to mount the rootfs partition and modules squashfs in there, which might have been the issue. 

There is no tty on default. You should try either using a serial debug cable (uart) or connecting Ethernet (it will dhcp automatically) and ssh in. 

If you build with the default skiff/core config it will start the Ubuntu desktop UI eventually (after downloading the container image on first startup). 

I'll take a look at adding a tty terminal on the default frame buffer output.

Anyway - seems it's not related to partition layout.

paralin, 2022-04-26:
@Lwing Could you please pull "master" up to latest commit - in commit 8e162b45705a7c15d56b2d4c83301d20be1f4cf5 I have enabled ext3 built in instead of a module, this might fix the issue for you. 

You just need to "git pull", then "make compile" once more and it will automatically re-pack the kernel & the image without rebuilding everything.

Lwing, 2022-04-26:
Interestingly, it boots a bit better - but not fully - after I got a new SD card. SSH is up, but yacht (at port 8000 according to the config) won't. The docker0 interface + firewall keeps going up and down even minute or so.. strange. 
Haven't tried the older commit (without the newest patch) with the new memory card.

paralin, 2022-04-26:
Probably an issue with yacht, can you ssh in and see?

Might also want to try apps/portainer.

Lwing, 2022-04-26:
Portainer works. Haven't had time to check anything further. Will check through SSH when I have the time. Thanks again for your help. The booting issue seems ok. Userspace I can debug on my own. I'll write when/if I find something interesting regarding yacht

Lwing, 2022-04-26:
I've found an issue with the build.
```
docker run   --name=yacht   --restart=always   -d -p 8000:8000   -v /var/run/docker.sock:/var/run/docker.sock   -v /mnt/persist/yaght-config:/config quay.io/skiffos/selfhostedpro-yacht:latest
WARNING: The requested image's platform (linux/arm64) does not match the detected host platform (linux/arm/v7) and no specific platform was requested
docker logs yacht
exec /init: exec format error
```
The container runs, but the app doesn't actually start. 

```
docker run   --name=yacht   --restart=always   -d -p 8000:8000   -v /var/run/docker.sock:/var/run/docker.sock   -v /mnt/persist/yaght-config:/config **selfhostedpro/yacht:latest**
```
works just fine. So either choose the correct image or rebuild system as 64bit?

paralin, 2022-04-26:
I need to re-build that image multi-arch. That is to say, it needs an arm64, arm, riscv, and amd64 version, then a manifest to link them. Will do that now.

paralin, 2022-04-26:
@Lwing Okay it's been updated & should work fine on arm, arm64, and amd64. Feel free to ping here & reopen if not.

closed 2022-04-26 by paralin

paralin, 2022-04-26:
Here it is running on the xu4:

[image]

(ARM)