skiffos/SkiffOS #202: Persist Partition Expansion not working

maxberger, 2022-05-17:
Hi,

on my build from this morning, the persist partition did not expand, and is thus already completely full. The SD card is a 32 GB card, so it should have plenty of space.

Is there any way to trigger the partition expansion manually?

Build: Welcome to SkiffOS 2022.02.1-175-g48750c7b!
Arch: allwinner/nezha

root@skiff-48750c7b:~# df -h
Filesystem      Size  Used Avail Use% Mounted on
overlay         1.0G   48K  1.0G   1% /
/dev/root       865M  849M     0 100% /mnt/persist
devtmpfs        242M     0  242M   0% /dev
tmpfs           243M     0  243M   0% /dev/shm
tmpfs            98M  576K   97M   1% /run
tmpfs           4.0M     0  4.0M   0% /sys/fs/cgroup
tmpfs           243M     0  243M   0% /tmp
overlay         865M  849M     0 100% /etc/NetworkManager/system-connections
/dev/loop1      8.2M  8.2M     0 100% /usr/lib/modules

paralin, 2022-05-17:
@maxberger please checkout latest `master` and try again, >= commit 612d689f8f18ce826c8f5ef5779b0b99be1f4c51

paralin, 2022-05-17:
@maxberger Interesting, I think the `resize2fs` step the `skiff-init-squashfs` does needs to resize the actual partition. I will need to update it slightly to fix this.

On most devices, `embiggen-disk` is used to resize the partition and then call `resize2fs`.

I'll probably release a patch tomorrow that uses it the same way in `skiff-init`.

In the meantime, you can resize the partition with `gparted` - just drag the end of the partition out to the end of the disk.

paralin, 2022-05-19:
@maxberger Please try on master, 62a5d82a57dc0d087e11a7919379e6ce550384cd or newer.

You'll need to "make compile" then run the "install" step once more.

If it fails to resize, you'll find the logs in "journalctl -u skiff-init" - please send them here.

Thanks!

maxberger, 2022-05-20:
I've rebuilt and rewritten the image to the SD card, and now my board doesn't boot up anymore, or at least not to the point where it would ask for an IP via DHCP. I will need to connect a serial console to find out more; I'll need a bit more time to set that up. I've also had bad SD cards in the past, so it may or may not be related to the latest changes.

paralin, 2022-05-21:
@maxberger Indeed I see a kernel panic on nezha, working on debugging it:

```
[    3.791577] CPU: 0 PID: 1 Comm: systemd Not tainted 5.18.0-rc7 #1
[    3.797672] Hardware name: Allwinner D1 Nezha (DT)
[    3.802460] epc : strscpy+0x2a/0xfa
[    3.805967]  ra : strscpy_pad+0x12/0x3a
[    3.809806] epc : ffffffff80311bb4 ra : ffffffff802e74a4 sp : ffffffc80400bbb0
[    3.817024]  gp : ffffffff81391a30 tp : ffffffd8017a0000 t0 : 00000000000001f8
[    3.824242]  t1 : ffffffc80400bbd0 t2 : 0000000000000000 s0 : ffffffd8030b3070
[    3.831459]  s1 : 0000000000000040 a0 : 0000000000000000 a1 : ffffffffffffffff
[    3.838676]  a2 : 0000000000000040 a3 : ffffffff80de30ca a4 : 0000000000000000
[    3.845892]  a5 : ffffffffffffffff a6 : ffffffffffffffff a7 : ffffffd8019152d0
[    3.853109]  s2 : ffffffc80400bbd0 s3 : 0000000000000000 s4 : 0000000000000000
[    3.860326]  s5 : 0000000000000058 s6 : 0000000000000001 s7 : 0000000000000000
[    3.867543]  s8 : 0000000000000000 s9 : 0000000000000002 s10: 0000000000000000
[    3.874759]  s11: 0000000000000001 t3 : 0000003fba14cfb8 t4 : 0000000000000016
[    3.881975]  t5 : 0000000000000061 t6 : ffffffd8017cb004
[    3.887283] status: 0000000200000120 badaddr: ffffffffffffffff cause: 000000000000000d
[    3.895194] [<ffffffff80311bb4>] strscpy+0x2a/0xfa
[    3.899994] [<ffffffff80185d80>] proc_task_name+0x2a/0x90
[    3.905398] [<ffffffff80141b42>] seq_put_decimal_ull_width+0x8e/0xdc
[    3.911764] [<ffffffff800335ac>] thread_group_cputime_adjusted+0x24/0x34
[    3.918476] [<ffffffff801860a6>] do_task_stat+0x2c0/0x8c8
[    3.923881] [<ffffffff800e0ff2>] kvmalloc_node+0x12/0xa8
[    3.929209] [<ffffffff80181c80>] proc_single_show+0x3e/0x88
[    3.934789] [<ffffffff801415d0>] seq_read_iter+0x130/0x2fa
[    3.940283] [<ffffffff8014181a>] seq_read+0x80/0x9e
[    3.945168] [<ffffffff801209ce>] vfs_read+0x5a/0x106
[    3.950143] [<ffffffff80120ccc>] ksys_read+0x38/0x98
[    3.955112] [<ffffffff80120d28>] ksys_read+0x94/0x98
[    3.960080] [<ffffffff8000232c>] check_syscall_nr+0x1e/0x22
[    3.965823] ---[ end trace 0000000000000000 ]---
[    3.970537] Kernel panic - not syncing: Attempted to kill init! exitcode=0x0000000b
[    3.978200] ---[ end Kernel panic - not syncing: Attempted to kill init! exitcode=0x0000000b ]---

```

paralin, 2022-05-21:
@maxberger I traced it to a buggy commit on 5.18.x and reverted it in  971d13bbb7e0e302afd533446141409068e30af9 - could you please rebuild your kernel by pulling latest & "rm -rf ./workspaces/default/build/linux-custom" then "make configure compile" ? thanks

maxberger, 2022-05-21:
Good news: The system boots again!

Bad news: Resize does not work, here is the output of journalctl: 

Looks like -ignore-resize-partition is not a valid argument.

```
May 21 14:05:26 skiff-012cc133 mount-all.sh[136]: Sat May 21 14:05:15 UTC 2022
May 21 14:05:26 skiff-012cc133 mount-all.sh[109]: Persist drive is at /mnt/persist path /mnt/persist
May 21 14:05:26 skiff-012cc133 mount-all.sh[109]: Configuring systemd-journald to use /mnt/persist/skiff/journal
May 21 14:05:26 skiff-012cc133 mount-all.sh[109]: Rootfs drive is at /mnt/rootfs.
May 21 14:05:26 skiff-012cc133 mount-all.sh[109]: Building SSH key list...
May 21 14:05:27 skiff-012cc133 mount-all.sh[175]: /etc/NetworkManager/system-connections is not a mountpoint
May 21 14:05:27 skiff-012cc133 mount-all.sh[109]: Resizing /mnt/persist with embiggen-disk...
May 21 14:05:27 skiff-012cc133 mount-all.sh[199]: flag provided but not defined: -ignore-resize-partition
May 21 14:05:27 skiff-012cc133 mount-all.sh[199]: Usage of embiggen-disk:
May 21 14:05:27 skiff-012cc133 mount-all.sh[199]: # embiggen-disk [flags] <mount-point-to-enlarge>
May 21 14:05:27 skiff-012cc133 mount-all.sh[199]:   -dry-run
May 21 14:05:27 skiff-012cc133 mount-all.sh[199]:             don't make changes
May 21 14:05:27 skiff-012cc133 mount-all.sh[199]:   -verbose
May 21 14:05:27 skiff-012cc133 mount-all.sh[199]:             verbose output
May 21 14:05:27 skiff-012cc133 mount-all.sh[109]: Failed to resize via mountpoint, trying via /dev/mmcblk0p1...
May 21 14:05:27 skiff-012cc133 mount-all.sh[202]: flag provided but not defined: -ignore-resize-partition
May 21 14:05:27 skiff-012cc133 mount-all.sh[202]: Usage of embiggen-disk:
May 21 14:05:27 skiff-012cc133 mount-all.sh[202]: # embiggen-disk [flags] <mount-point-to-enlarge>
May 21 14:05:27 skiff-012cc133 mount-all.sh[202]:   -dry-run
May 21 14:05:27 skiff-012cc133 mount-all.sh[202]:             don't make changes
May 21 14:05:27 skiff-012cc133 mount-all.sh[202]:   -verbose
May 21 14:05:27 skiff-012cc133 mount-all.sh[202]:             verbose output
May 21 14:05:27 skiff-012cc133 mount-all.sh[109]: Failed to resize persist. Continuing anyway.
May 21 14:05:27 skiff-012cc133 mount-all.sh[205]: Configuring Docker to use /mnt/persist/skiff/docker
May 21 14:05:27 skiff-012cc133 mount-all.sh[205]: Configuring Docker to use systemd-journald
May 21 14:05:27 skiff-012cc133 mount-all.sh[205]: Configuring Docker to start with '/usr/bin/dockerd -H fd:// --data-rooMay 21 14:05:27 skiff-012cc133 mount-all.sh[216]: /usr/lib/modules is not a mountpoint
May 21 14:05:27 skiff-012cc133 mount-all.sh[215]: Mounting modules from squashfs...
```

Just to make sure the resizing works, I ran the commands manually (without the -ignore...):
```
root@skiff-012cc133:~# embiggen-disk /mnt/persist
Changes made:
  * partition /dev/mmcblk0p1: before: 1835008 sectors, after: 62069760 sectors
error: running /usr/sbin/resize2fs [resize2fs /dev/mmcblk0p1]: exit status 1, resize2fs 1.46.5 (30-Dec-2021)
resize2fs: Device or resource busy while trying to open /dev/mmcblk0p1
Couldn't find valid filesystem superblock.
```
then `sync`, `reboot`, and now I have space! So manual works, but automatic does not. As said, since this is an experimental system I am more than willing to test this again once you have a newer version. 

Thank you for all you support and for this OS!

paralin, 2022-05-21:
@maxberger it is a valid argument, but added recently-

Please run "git submodule update" and then "rm -rf workspaces/*/build/embiggen-disk*" to force rebuilding the embiggen disk tool, then "make compile" 

That should fix that error.

maxberger, 2022-05-21:
Unfortunately that still did not do the trick :(

I had to do a full make clean / rebuild, because updating gave me a strange error with glibc complaining about kernel header files. Once that was done, I was able to create the disk, and it booted -> Just to be stuck on a 1GB persist partition again.

log file for first new boot:
```
May 21 21:08:49 skiff-012cc133 mount-all.sh[107]: Build date of Sat May 21 23:08:33 CEST 2022 is newer than internal May 21 21:08:49 skiff-012cc133 mount-all.sh[130]: Sat May 21 21:08:33 UTC 2022
May 21 21:08:49 skiff-012cc133 mount-all.sh[107]: Persist drive is at /mnt/persist path /mnt/persist
May 21 21:08:49 skiff-012cc133 mount-all.sh[107]: Configuring systemd-journald to use /mnt/persist/skiff/journal
May 21 21:08:49 skiff-012cc133 mount-all.sh[107]: Rootfs drive is at /mnt/rootfs.
May 21 21:08:49 skiff-012cc133 mount-all.sh[107]: Building SSH key list...
May 21 21:08:50 skiff-012cc133 mount-all.sh[169]: /etc/NetworkManager/system-connections is not a mountpoint
May 21 21:08:50 skiff-012cc133 mount-all.sh[107]: Resizing /mnt/persist with embiggen-disk...
May 21 21:08:50 skiff-012cc133 mount-all.sh[211]: 2022/05/21 21:08:42 statFSFindmnt: running(/usr/bin/findmnt --noheaMay 21 21:08:50 skiff-012cc133 mount-all.sh[211]: 2022/05/21 21:08:42 getFileSystemResizer("/mnt/persist") = main.fsRMay 21 21:08:50 skiff-012cc133 mount-all.sh[211]: 2022/05/21 21:08:42 statFSFindmnt: running(/usr/bin/findmnt --noheaMay 21 21:08:50 skiff-012cc133 mount-all.sh[211]: 2022/05/21 21:08:42 fsResizer.DepResizer: returning partitionResizeMay 21 21:08:50 skiff-012cc133 mount-all.sh[211]: 2022/05/21 21:08:42 Resizing partition "/dev/mmcblk0p1" ...
May 21 21:08:50 skiff-012cc133 mount-all.sh[211]: 2022/05/21 21:08:42 Getting partition table for "/dev/mmcblk0" ...
May 21 21:08:50 skiff-012cc133 mount-all.sh[211]: 2022/05/21 21:08:42 Device "/dev/mmcblk0" has 1 partitions.
May 21 21:08:50 skiff-012cc133 mount-all.sh[211]: Current partition table:
May 21 21:08:50 skiff-012cc133 mount-all.sh[211]: label: dos
May 21 21:08:50 skiff-012cc133 mount-all.sh[211]: label-id: 0x458cced4
May 21 21:08:50 skiff-012cc133 mount-all.sh[211]: device: /dev/mmcblk0
May 21 21:08:50 skiff-012cc133 mount-all.sh[211]: unit: sectors
May 21 21:08:50 skiff-012cc133 mount-all.sh[211]: sector-size: 512
May 21 21:08:50 skiff-012cc133 mount-all.sh[211]: /dev/mmcblk0p1 : start=262144, size=1835008, type=83
May 21 21:08:50 skiff-012cc133 mount-all.sh[211]: Cur size: 62333952
May 21 21:08:50 skiff-012cc133 mount-all.sh[211]: Part start: 262144
May 21 21:08:50 skiff-012cc133 mount-all.sh[211]: Part size: 1835008
May 21 21:08:50 skiff-012cc133 mount-all.sh[211]: Part end: 2097152
May 21 21:08:50 skiff-012cc133 mount-all.sh[211]: Remaining after final partition: 60236800
May 21 21:08:50 skiff-012cc133 mount-all.sh[211]: Need to extend disk by 60234752 sectors (30840193024 bytes, 28.722
May 21 21:08:50 skiff-012cc133 mount-all.sh[211]: New partition table to write:
May 21 21:08:50 skiff-012cc133 mount-all.sh[211]: label: dos
May 21 21:08:50 skiff-012cc133 mount-all.sh[211]: label-id: 0x458cced4
May 21 21:08:50 skiff-012cc133 mount-all.sh[211]: device: /dev/mmcblk0
May 21 21:08:50 skiff-012cc133 mount-all.sh[211]: unit: sectors
May 21 21:08:50 skiff-012cc133 mount-all.sh[211]: sector-size: 512
May 21 21:08:50 skiff-012cc133 mount-all.sh[211]: /dev/mmcblk0p1 : start=262144, size=62069760, type=83
May 21 21:08:50 skiff-012cc133 mount-all.sh[211]: Setting new partition table...
May 21 21:08:52 skiff-012cc133 mount-all.sh[218]: Disk /dev/mmcblk0: 29.72 GiB, 31914983424 bytes, 62333952 sectors
May 21 21:08:52 skiff-012cc133 mount-all.sh[218]: Units: sectors of 1 * 512 = 512 bytes
May 21 21:08:52 skiff-012cc133 mount-all.sh[218]: Sector size (logical/physical): 512 bytes / 512 bytes
May 21 21:08:52 skiff-012cc133 mount-all.sh[218]: I/O size (minimum/optimal): 512 bytes / 512 bytes
May 21 21:08:52 skiff-012cc133 mount-all.sh[218]: Disklabel type: dos
May 21 21:08:52 skiff-012cc133 mount-all.sh[218]: Disk identifier: 0x458cced4
May 21 21:08:52 skiff-012cc133 mount-all.sh[218]: Old situation:
May 21 21:08:52 skiff-012cc133 mount-all.sh[218]: Device         Boot  Start     End Sectors  Size Id Type
May 21 21:08:52 skiff-012cc133 mount-all.sh[218]: /dev/mmcblk0p1      262144 2097151 1835008  896M 83 Linux
May 21 21:08:52 skiff-012cc133 mount-all.sh[218]: >>> Script header accepted.
May 21 21:08:52 skiff-012cc133 mount-all.sh[218]: >>> Script header accepted.
May 21 21:08:52 skiff-012cc133 mount-all.sh[218]: >>> Script header accepted.
May 21 21:08:52 skiff-012cc133 mount-all.sh[218]: >>> Script header accepted.
May 21 21:08:52 skiff-012cc133 mount-all.sh[218]: >>> Script header accepted.
May 21 21:08:52 skiff-012cc133 mount-all.sh[218]: >>> Created a new DOS disklabel with disk identifier 0x458cced4.
```

since the log looked correct, I rebooted (sync, reboot); this changed the log but the persist partition is still stuck at 1GB.

log file for 2nd boot (3rd boot has the same output):
```
May 21 21:08:49 skiff-012cc133 mount-all.sh[108]: Build date of Sat May 21 23:08:33 CEST 2022 is newer than internal clock of Fri Mar 11 07:33:35 UTC 2022. Bumping internal clock to build date.
May 21 21:08:49 skiff-012cc133 mount-all.sh[131]: Sat May 21 21:08:33 UTC 2022
May 21 21:08:49 skiff-012cc133 mount-all.sh[108]: Persist drive is at /mnt/persist path /mnt/persist
[... cut ...]
May 21 21:08:50 skiff-012cc133 mount-all.sh[221]: error: running /usr/sbin/resize2fs [resize2fs /dev/mmcblk0p1]: exit status 1, resize2fs 1.46.5 (30-Dec-2021)
May 21 21:08:50 skiff-012cc133 mount-all.sh[221]: resize2fs: Device or resource busy while trying to open /dev/mmcblk0p1
May 21 21:08:50 skiff-012cc133 mount-all.sh[221]: Couldn't find valid filesystem superblock.
May 21 21:08:50 skiff-012cc133 mount-all.sh[108]: Failed to resize persist. Continuing anyway.
May 21 21:08:50 skiff-012cc133 mount-all.sh[230]: /usr/lib/modules is not a mountpoint
May 21 21:08:50 skiff-012cc133 mount-all.sh[229]: Mounting modules from squashfs...
```

paralin, 2022-05-21:
I guess resize2fs doesn't like that it's already mounted. It should detect that it's mounted already but I don't think it will because of how skiff init squashfs works. 

The fix is probably just to mount the disk somewhere else so that resize2fs realizes online resizing is required. 

I'll test it today

paralin, 2022-05-22:
To summarize the issue:

 - resize2fs: thinks that /dev/mmcblk0p1 is not mounted
   - this is because the mount is done from outside the chroot
 - tries to open /dev/mmcblk0p1 (offline resize)
 - fails to open that

So to fix this, a workaround is to mount /dev/mmcblk0p1 to a different location so that resize2fs understands it is mounted, and tries to do the online resize instead.

paralin, 2022-05-22:
Fixed in 961b4fc - tested with nezha - resizes the fs properly.

```bash
        # Sometimes the path does not appear in the mounts list.
        #
        # Workaround: temporarily mount it so resize2fs realizes it needs to
        # do an online resize. Otherwise returns an error.
        ROOT_MOUNT_SOURCE=$(findmnt --target / --first-only -n -o "SOURCE")
        if [ -n "${FORCE_RESIZE_PERSIST_MOUNT}" ] || \
               [[ "${ROOT_MOUNT_SOURCE}" != "${PERSIST_DEVICE}" ]] || \
               ! findmnt --source ${PERSIST_DEVICE} --first-only; then
            echo "Temporarily mounting ${PERSIST_DEVICE} to force online resize..."
            RESIZE_DISK_MTPT=/mnt/.persist-resize
            mkdir -p ${RESIZE_DISK_MTPT}
            mount ${PERSIST_DEVICE} ${RESIZE_DISK_MTPT} || true
        fi
        echo "Resizing with embiggen-disk via ${PERSIST_DEVICE}..."
        if ${RESIZE_DISK[@]} ${PERSIST_DEVICE}; then
            PERSIST_RESIZED="true"
        fi
        if [ -n "${RESIZE_DISK_MTPT}" ]; then
            echo "Unmounting ${RESIZE_DISK_MTPT}..."
            umount ${RESIZE_DISK_MTPT} || true
        fi
    fi
``

closed 2022-05-22 by paralin

maxberger, 2022-05-22:
I can confirm it worked, thank you so much!

commit 961b4fc4 (2022-05-22): configs-base: force resize2fs to perform online resize of /

Fixes #202