skiffos/SkiffOS #206: fails to build with rpi/4,core/alpine using build/Docker: undefined reference to `__lll_lock_wake_private'

conversy, 2022-05-19:
logs:

```
/home/buildroot/skiff/workspaces/default/host/bin/aarch64-buildroot-linux-gnu-gcc   -nostdlib -nostartfiles -shared -o /home/buildroot/skiff/workspaces/default/build/glibc-2.35-96-g2c4fc8e5ca742c6a3a1933799495bb0b00a807f0/build/elf/ld.so.new		\
	  -Wl,-z,combreloc -Wl,-z,relro -Wl,--hash-style=both -Wl,-z,defs 	\
	  /home/buildroot/skiff/workspaces/default/build/glibc-2.35-96-g2c4fc8e5ca742c6a3a1933799495bb0b00a807f0/build/elf/librtld.os -Wl,--version-script=/home/buildroot/skiff/workspaces/default/build/glibc-2.35-96-g2c4fc8e5ca742c6a3a1933799495bb0b00a807f0/build/ld.map		\
	  -Wl,-soname=ld-linux-aarch64.so.1			\
	  -Wl,-defsym=_begin=0
/home/buildroot/skiff/workspaces/default/host/lib/gcc/aarch64-buildroot-linux-gnu/11.3.0/../../../../aarch64-buildroot-linux-gnu/bin/ld: /home/buildroot/skiff/workspaces/default/build/glibc-2.35-96-g2c4fc8e5ca742c6a3a1933799495bb0b00a807f0/build/elf/librtld.os: in function `_dl_make_stacks_executable':
(.text+0x2690): undefined reference to `__lll_lock_wake_private'
/home/buildroot/skiff/workspaces/default/host/lib/gcc/aarch64-buildroot-linux-gnu/11.3.0/../../../../aarch64-buildroot-linux-gnu/bin/ld: (.text+0x26a0): undefined reference to `__lll_lock_wait_private'
/home/buildroot/skiff/workspaces/default/host/lib/gcc/aarch64-buildroot-linux-gnu/11.3.0/../../../../aarch64-buildroot-linux-gnu/bin/ld: /home/buildroot/skiff/workspaces/default/build/glibc-2.35-96-g2c4fc8e5ca742c6a3a1933799495bb0b00a807f0/build/elf/librtld.os: in function `__thread_gscope_wait':
(.text+0xee24): undefined reference to `__lll_lock_wait_private'
/home/buildroot/skiff/workspaces/default/host/lib/gcc/aarch64-buildroot-linux-gnu/11.3.0/../../../../aarch64-buildroot-linux-gnu/bin/ld: (.text+0xee44): undefined reference to `__lll_lock_wake_private'
/home/buildroot/skiff/workspaces/default/host/lib/gcc/aarch64-buildroot-linux-gnu/11.3.0/../../../../aarch64-buildroot-linux-gnu/bin/ld: /home/buildroot/skiff/workspaces/default/build/glibc-2.35-96-g2c4fc8e5ca742c6a3a1933799495bb0b00a807f0/build/elf/librtld.os: in function `_dl_init_static_tls':
(.text+0xff14): undefined reference to `__lll_lock_wait_private'
/home/buildroot/skiff/workspaces/default/host/lib/gcc/aarch64-buildroot-linux-gnu/11.3.0/../../../../aarch64-buildroot-linux-gnu/bin/ld: (.text+0xff30): undefined reference to `__lll_lock_wake_private'
collect2: error: ld returned 1 exit status
make[6]: *** [Makefile:1291: /home/buildroot/skiff/workspaces/default/build/glibc-2.35-96-g2c4fc8e5ca742c6a3a1933799495bb0b00a807f0/build/elf/ld.so] Error 1
make[5]: *** [Makefile:483: elf/subdir_lib] Error 2
make[4]: *** [Makefile:9: all] Error 2
make[3]: *** [package/pkg-generic.mk:293: /home/buildroot/skiff/workspaces/default/build/glibc-2.35-96-g2c4fc8e5ca742c6a3a1933799495bb0b00a807f0/.stamp_built] Error 2
make[2]: *** [Makefile:23: _all] Error 2
make[2]: Leaving directory '/home/buildroot/skiff/workspaces/default'
make[1]: *** [Makefile:64: compile] Error 2
make[1]: Leaving directory '/home/buildroot/skiff/build'
make: *** [Makefile:2: compile] Error 2
```

paralin, 2022-05-19:
@conversy Please "make clean" then "make compile" again, I think your glibc and/or gcc build are probably broken somehow.

conversy, 2022-05-21:
FYI I started from scratch, so I guess a `make clean` might be unnecessary. Anyway I did as you suggested, and encountered another problem (see below). I did it with `export SKIFF_CONFIG=pi/4,core/alpine` though, I might have misunderstood something?

```
cd /home/buildroot/skiff/workspaces/default/build/glibc-2.35-96-g2c4fc8e5ca742c6a3a1933799495bb0b00a807f0/build && /home/buildroot/skiff/workspaces/default/host/lib/gcc/aarch64-buildroot-linux-gnu/11.3.0/../../../../aarch64-buildroot-linux-gnu/bin/ar cruv libc_nonshared.a `cat csu/stamp.oS iconv/stamp.oS locale/stamp.oS localedata/stamp.oS iconvdata/stamp.oS assert/stamp.oS ctype/stamp.oS intl/stamp.oS catgets/stamp.oS math/stamp.oS setjmp/stamp.oS signal/stamp.oS stdlib/stamp.oS stdio-common/stamp.oS libio/stamp.oS dlfcn/stamp.oS nptl/stamp.oS rt/stamp.oS malloc/stamp.oS string/stamp.oS wcsmbs/stamp.oS timezone/stamp.oS time/stamp.oS dirent/stamp.oS grp/stamp.oS pwd/stamp.oS posix/stamp.oS io/stamp.oS termios/stamp.oS resource/stamp.oS misc/stamp.oS socket/stamp.oS sysvipc/stamp.oS gmon/stamp.oS gnulib/stamp.oS wctype/stamp.oS manual/stamp.oS shadow/stamp.oS gshadow/stamp.oS po/stamp.oS argp/stamp.oS conform/stamp.oS debug/stamp.oS mathvec/stamp.oS support/stamp.oS crypt/stamp.oS nptl_db/stamp.oS inet/stamp.oS resolv/stamp.oS nss/stamp.oS hesiod/stamp.oS sunrpc/stamp.oS nis/stamp.oS nscd/stamp.oS login/stamp.oS elf/stamp.oS stamp.oS`
touch /home/buildroot/skiff/workspaces/default/build/glibc-2.35-96-g2c4fc8e5ca742c6a3a1933799495bb0b00a807f0/build/runtime-linker.stamp
/home/buildroot/skiff/workspaces/default/host/lib/gcc/aarch64-buildroot-linux-gnu/11.3.0/../../../../aarch64-buildroot-linux-gnu/bin/ar: cilacicmssstdlib/atexit.oS: No such file or directory
make[6]: *** [../o-iterator.mk:9: /home/buildroot/skiff/workspaces/default/build/glibc-2.35-96-g2c4fc8e5ca742c6a3a1933799495bb0b00a807f0/build/libc_nonshared.a] Error 1
```

paralin, 2022-05-21:
@conversy These are weird compiler errors, I can't reproduce them so probably either something is broken with your host system you're trying to build on, or something is messed up in the compiler cache.

Please try this -

```
rm -rf ./workspaces/default ./workspaces/.ccache/default
make configure compile
```

That will clear out the cache.

conversy, 2022-05-22:
I'm building it on my mac (m1), but I thought it was not important as it was using docker... Could it explain this behavior? I'll try to clear out the cache anyway.

paralin, 2022-05-22:
It should work, but I'm not sure if anyone has tested the full buildroot build on a m1 cpu yet. 

If you're not doing so already, I'd recommend trying to use Lima (Linux on Mac) to run Linux & build in there: https://github.com/lima-vm/lima

Thanks for testing, if you get it working, it will be the first successful build on M1, as far as I know!

paralin, 2022-06-03:
Closing for now but feel free to msg here + reopen if needed.

closed 2022-06-03 by paralin