skiffos/SkiffOS #208: error compiling embiggen-disk

rcpehls, 2022-05-21:
Skiff config matches apps/docker,skiff/core,jetson/common,jetson/nano, skipping config rebuild.
Kicking off Buildroot build...
cd $BUILDROOT_DIR && make 
make[2]: Entering directory '/home/sirros/SkiffOS/workspaces/default'
ERROR: embiggen-disk-v20220519.tar.gz has wrong sha256 hash:
ERROR: expected: ed519f8c7b70136c96eed13571c14d344fc31f57e11aa409f72e885da37b300e
ERROR: got     : 6c1896d89691763fc0a9696a56e54a3e66730f35dedb2add7bd1cda8e9037cb1
ERROR: Incomplete download, or man-in-the-middle (MITM) attack
dl-wrapper: Re-downloading 'embiggen-disk-v20220519.tar.gz'...
wget --passive-ftp -nd -t 3 -O '/home/sirros/SkiffOS/workspaces/default/build/.embiggen-disk-v20220519.tar.gz.ePDT11/output' 'https://github.com/skiffos/embiggen-disk/archive/v20220519/embiggen-disk-v20220519.tar.gz' 
--2022-05-21 17:59:07--  https://github.com/skiffos/embiggen-disk/archive/v20220519/embiggen-disk-v20220519.tar.gz
Resolving github.com (github.com)... 20.201.28.151
Connecting to github.com (github.com)|20.201.28.151|:443... connected.
HTTP request sent, awaiting response... 302 Found
Location: https://codeload.github.com/skiffos/embiggen-disk/tar.gz/refs/tags/v20220519 [following]
--2022-05-21 17:59:08--  https://codeload.github.com/skiffos/embiggen-disk/tar.gz/refs/tags/v20220519
Resolving codeload.github.com (codeload.github.com)... 20.201.28.149
Connecting to codeload.github.com (codeload.github.com)|20.201.28.149|:443... connected.
HTTP request sent, awaiting response... 200 OK
Length: unspecified [application/x-gzip]
Saving to: '/home/sirros/SkiffOS/workspaces/default/build/.embiggen-disk-v20220519.tar.gz.ePDT11/output'

/home/sirros/SkiffOS/workspaces/default     [ <=>                                                                          ]  31.20K  --.-KB/s    in 0.02s   

2022-05-21 17:59:08 (1.38 MB/s) - '/home/sirros/SkiffOS/workspaces/default/build/.embiggen-disk-v20220519.tar.gz.ePDT11/output' saved [31949]

# github.com/pquerna/ffjson v0.0.0-20190930134022-aa0246cd15f7
## explicit
github.com/pquerna/ffjson/fflib/v1
github.com/pquerna/ffjson/fflib/v1/internal
# github.com/u-root/u-root v0.0.0-20180806213625-12f9029297cf
## explicit
github.com/u-root/u-root/pkg/cpio
github.com/u-root/u-root/pkg/uio
# golang.org/x/sys v0.0.0-20201112073958-5cba982894dd
## explicit
golang.org/x/sys/internal/unsafeheader
golang.org/x/sys/unix
ERROR: embiggen-disk-v20220519.tar.gz has wrong sha256 hash:
ERROR: expected: ed519f8c7b70136c96eed13571c14d344fc31f57e11aa409f72e885da37b300e
ERROR: got     : 6c1896d89691763fc0a9696a56e54a3e66730f35dedb2add7bd1cda8e9037cb1
ERROR: Incomplete download, or man-in-the-middle (MITM) attack
wget --passive-ftp -nd -t 3 -O '/home/sirros/SkiffOS/workspaces/default/build/.embiggen-disk-v20220519.tar.gz.zejPm9/output' 'https://github.com/skiffos/mirror/raw/master/embiggen-disk/embiggen-disk-v20220519.tar.gz' 
--2022-05-21 17:59:08--  https://github.com/skiffos/mirror/raw/master/embiggen-disk/embiggen-disk-v20220519.tar.gz
Resolving github.com (github.com)... 20.201.28.151
Connecting to github.com (github.com)|20.201.28.151|:443... connected.
HTTP request sent, awaiting response... 404 Not Found
2022-05-21 17:59:09 ERROR 404: Not Found.

wget --passive-ftp -nd -t 3 -O '/home/sirros/SkiffOS/workspaces/default/build/.embiggen-disk-v20220519.tar.gz.0pmgwz/output' 'https://github.com/skiffos/mirror/raw/master/embiggen-disk-v20220519.tar.gz' 
--2022-05-21 17:59:09--  https://github.com/skiffos/mirror/raw/master/embiggen-disk-v20220519.tar.gz
Resolving github.com (github.com)... 20.201.28.151
Connecting to github.com (github.com)|20.201.28.151|:443... connected.
HTTP request sent, awaiting response... 404 Not Found
2022-05-21 17:59:09 ERROR 404: Not Found.

package/pkg-generic.mk:185: recipe for target '/home/sirros/SkiffOS/workspaces/default/build/embiggen-disk-v20220519/.stamp_downloaded' failed
make[3]: *** [/home/sirros/SkiffOS/workspaces/default/build/embiggen-disk-v20220519/.stamp_downloaded] Error 1
Makefile:23: recipe for target '_all' failed
make[2]: *** [_all] Error 2
make[2]: Leaving directory '/home/sirros/SkiffOS/workspaces/default'
Makefile:63: recipe for target 'compile' failed
make[1]: *** [compile] Error 2
make[1]: Leaving directory '/home/sirros/SkiffOS/build'
Makefile:2: recipe for target 'compile' failed
make: *** [compile] Error 2

paralin, 2022-05-21:
Will have a look thanks for the report.

paralin, 2022-05-21:
Please run "git submodule update" on the latest master revision and try.

paralin, 2022-05-21:
@rcpehls OK, I also added it to the mirror, so it should 100% work this time even if the go module checksum check fails.

closed 2022-05-21 by paralin

paralin, 2022-09-13:
Hi @rcpehls the Jetson AGX Orin and Xavier are now supported with **jetson/agx** - in case that's useful to you!