skiffos/SkiffOS #209: Pinebook support

yeyeto2788, 2022-05-25:
Hey guys! Awesome work here.

I'd like to use SkiffOS on the [Pinebook](https://wiki.pine64.org/wiki/1080P_Pinebook) and I'm more than happy to help on the testing side of things. 

I wanted to give a try but I don't know where to start rather than just creating the ticket and from the configs I don't see the A64 SoC within the Allwinner or Pine64 namespaces.

Best.

yeyeto2788, 2022-05-25:
Hey @paralin! Thanks for the reply but isn't that for the Pinebook Pro? I own the non Pro version which has a different SoC.

paralin, 2022-05-25:
Pro uses rk3399 which is a 4x 1.5 GHz Cortex-A53 & 2x 2Ghz Cortex-A72

Original uses the allwinner a64

So yes they are different - I could have sworn I added the a64 for a
different board, maybe it was the rockpro?

Anyway, I'll have a look at adding a config for the pinebook 1.

yeyeto2788, 2022-05-25:
Thanks! If you want me to test it out just let me know. 

I would love also to know how to port this into any device as I also have other Orange PI SBC.

paralin, 2022-05-25:
Feel free to open one issue per board you want added.

It's not much work, if you have a look at any of the existing configs/ they are all plaintext files.

paralin, 2022-05-31:
@yeyeto2788 Added PR #210 - if you could compile the `pinebook` branch with `pine64/book_a64` and see if that works?

it might not work because u-boot needs to be re-built for the pinebook (re-using the pinephone a64 one for now) - but then again, we might be able to get away with reusing those blobs, so let's give it a shot.

There's not much info about the pinebook given its obsolescence.

yeyeto2788, 2022-05-31:
Building it right now 🥳 

I found some information at the following sites but I guess you've also seen them already.
* [SoC](https://linux-sunxi.org/A64)
* [PineTab](https://linux-sunxi.org/PineTab) which should also be almost the same
* [Armbian documentation](https://docs.armbian.com/board_details/pine64/) and the [download image](https://www.armbian.com/pinebook-a64/) for Armbian

Once it's built and tested I'll update the ticket. Thanks!

paralin, 2022-05-31:
@yeyeto2788 I accidentally updated the pinebook pro config instead of the a64 one - updated the PR to fix it, now book_a64 is configured correctly, I think. 

You might need to force rebuilding the firmware / uboot related packages for it to work correctly,

Sorry about that mistake!

paralin, 2022-06-01:
PR: #210 is ready for merge

paralin, 2022-06-03:
Merged, with the kernel config switched to arch defconfig. Looking forward to hearing your test results. If there is anything that needs fixing still I'll re-open this.

paralin, 2022-06-03:
Reopening to address two issues:

 - Display not working
 - USB ethernet not working

paralin, 2022-08-07:
@yeyeto2788 USB ethernet might work now.

paralin, 2022-10-26:
@yeyeto2788 Hopefully this fixes it: https://github.com/skiffos/SkiffOS/commit/eee57b1187c0e2160a353fcef4e4535eece7c135

Awaiting your testing!

closed 2024-11-24 by paralin