skiffos/SkiffOS #216: Odroid M1 with eMMC?

peterwilli, 2022-07-14:
Hey there, I wondered if it was possible to complete the Odroid M1 tutorial (https://github.com/skiffos/SkiffOS/tree/master/configs/odroid/m1) on eMMC since the modules are faster especially if you interact a lot with the persist partition.

I know that SkiffOS lives in RAM, so it hardly matters, but eMMC doesn't mount if you're booting from SD, at least I can't get SkiffOS (nor any other Linux distro supported on this SBC) to recognize it. Thanks!

peterwilli, 2022-07-15:
> Checking the logs on the serial debug port would help in this case.

How can I do this? I haven't done that before! I'd love to help out here.

paralin, 2022-07-15:
@peterwilli You need a UART cable:

[image]

Connect it to the port on the board that fits the port,

then on your laptop (linux, I hope?)

```
sudo screen /dev/ttyUSB0 115200
```

... assuming it went to `/dev/ttyUSB0`

or you could use `minicom -D /dev/ttyUSB0 -b 115200`

peterwilli, 2022-07-15:
> then on your laptop (linux, I hope?)

Always!

I sadly don't have a cable like that. Isn't there a way to output it over the HDMI display cable? I always used that, but on SkiffOS it's completely blank somehow.

I can get a cable like that if no other solution can work...

paralin, 2022-07-15:
@peterwilli well, this is for debugging u-boot, for booting from the EMMC. u-boot doesn't output to HDMI, only the serial port.

Here are a few places you can get one:

 - https://www.hardkernel.com/shop/usb-uart-module-kit/
 - https://www.amazon.com/ODROID-0-USB-UART-Module-Kit/dp/B01LEV2OWA
 - https://ameridroid.com/products/usb-uart-module-kit

peterwilli, 2022-07-15:
> @peterwilli well, this is for debugging u-boot, for booting from the EMMC. u-boot doesn't output to HDMI, only the serial port.
> 
> Here are a few places you can get one:
> 
>     * https://www.hardkernel.com/shop/usb-uart-module-kit/
> 
>     * https://www.amazon.com/ODROID-0-USB-UART-Module-Kit/dp/B01LEV2OWA
> 
>     * https://ameridroid.com/products/usb-uart-module-kit
> 
> 
> ... if you can't get one, no problem, i'll have some time to test this soon and make sure it works.

I'll buy one today if I can't get it to run with Arduino because it's probably faster to build one myself with Arduino (done it before) than to wait for one to ship. On the other hand, I don't have a build lying around anymore, so maybe it's better to buy one online anyway as it takes time before it's completed.

peterwilli, 2022-07-15:
I managed to buy this. I think it'll be compatible as the chip is the same as on the Odroid version, it's cheaper and in stock and can be shipped tomorrow so that's nice as well.

[image]

paralin, 2022-07-18:
@peterwilli baudrate is wrong, or there are multiple instances of screen running. 

Maybe try "sudo killall -9 screen"

Then "screen /dev/ttyUSB0 115200"

Or try the higher baudrate from the wiki

screen /dev/ttyUSB0 1500000

peterwilli, 2022-07-19:
@paralin the eMMC works now! I just tried your update and it's there. 

```
root@skiff-a48e7829:~# lsblk
NAME         MAJ:MIN RM   SIZE RO TYPE MOUNTPOINTS
loop0          7:0    0  39.2M  0 loop /usr/lib/modules
mmcblk1      179:0    0 119.4G  0 disk 
|-mmcblk1p1  179:1    0     2G  0 part /mnt/rootfs
|                                      /mnt/boot
`-mmcblk1p2  179:2    0 117.4G  0 part /etc/ssh
                                       /var/log/journal
                                       /mnt/persist
mmcblk0      179:32   0  58.2G  0 disk 
|-mmcblk0p1  179:33   0   256M  0 part 
`-mmcblk0p2  179:34   0    58G  0 part 
mmcblk0boot0 179:64   0     4M  1 disk 
mmcblk0boot1 179:96   0     4M  1 disk 
zram0        253:0    0     2G  0 disk [SWAP]
```

The `mmcblk0` module is the eMMC. I'll try to boot from it now, but this is already great.

Thanks so much for helping out in both cases. If you want, I can still try the serial thingy, but I doubt we see any error now :laughing:

peterwilli, 2022-07-19:
> Hi, it should boot fine if you flash everything to the EMMC as if it was the SD card.
> 
> But; you say you tried this and it didn't work, while using a USB adapter does work.
> 
> Checking the logs on the serial debug port would help in this case.
> 
> One way you could workaround this in the short term:
> 
>     * Flash the SD card but delete the "persist" partition after flashing.
> 
>     * Create a partition labeled "persist" on the emmc.
> 
>     * Connect both the SD card and Emmc.
> 
>     * Skiff will use the "persist" labeled partition for storing data.
> 
> 
> Use the `msdos` partition type (not gpt).
> 
> When the EMMC is connected to the board you should be able to see it at /dev/mmcblk1 or similar. If not, there might be something preventing the kernel from seeing the emmc (as you've mentioned).

I tried this, and while my eMMC is now recognized, the persist partition isn't automatically mounted...

My lsblk:
```
root@skiff-a48e7829:~# lsblk
NAME         MAJ:MIN RM  SIZE RO TYPE MOUNTPOINTS
loop0          7:0    0 39.9M  0 loop /usr/lib/modules
mmcblk1      179:0    0 14.8G  0 disk 
`-mmcblk1p1  179:1    0    2G  0 part /mnt/rootfs
                                      /mnt/boot
mmcblk0      179:32   0 58.2G  0 disk 
`-mmcblk0p1  179:33   0 58.2G  0 part 
mmcblk0boot0 179:64   0    4M  1 disk 
mmcblk0boot1 179:96   0    4M  1 disk 
zram0        253:0    0    2G  0 disk [SWAP]
```

Gparted of SD card:
[image]

Gparted of eMMC:
[image]

peterwilli, 2022-07-19:
> @peterwilli the boot process is separate from core and all the other application level stuff. It's contained purely in the configs-base and the device packages.
> 
> Will try it out with my M1 and see what needs to be changed to make it work

Thanks, I'll be awaiting your response, and thanks for explaining!

paralin, 2022-07-19:
Ok, I traced it, the EMMC driver needed to be compiled built-in instead of as a module; this is because the modules are not available until after the EMMC is accessed (and persist is mounted).

```
CONFIG_MMC_SDHCI_OF_DWCMSHC=y
```

Fixed: 925dcecc768948deff4a57221590db759aa47240

Please `git pull`, `rm -rf workspaces/default/build/linux-custom` then `make compile` and run the `install` command again to update the files on the SD card.

peterwilli, 2022-07-20:
> @peterwilli Ok, I traced it, the EMMC driver needed to be compiled built-in instead of as a module; this is because the modules are not available until after the EMMC is accessed (and persist is mounted).
> 
> ```
> CONFIG_MMC_SDHCI_OF_DWCMSHC=y
> ```
> 
> Fixed: [925dcec](https://github.com/skiffos/SkiffOS/commit/925dcecc768948deff4a57221590db759aa47240)
> 
> Please `git pull`, `rm -rf workspaces/default/build/linux-custom` then `make compile` and run the `install` command again to update the files on the SD card.
> 
> Keep in mind: the first boot, it will need to allocate the swapfile on the emmc, which takes ~30 seconds before Docker starts.

I'm super grateful, I can't promise I can try it today due to getting my wisdom tooth removed, and I have no idea how I am afterward, but I'll definitively aim to try this today. Also, thanks for letting me know what the issue is and how you solved it, this really helps to see how things work, and maybe figure things out myself as well.

peterwilli, 2022-07-20:
Alright, I think we can close this. Not only does SD work, but booting from eMMC entirely works like a charm! Thanks so much!

closed 2022-07-20 by peterwilli